fkie-cad/FACT_core #23: Report generation

plutec, 2017-11-08:
It would be great if exists the posibility to create a report (HTML, for instance) given a firmware with all the information extracted.

dorpvom, 2017-11-08:
Hi,
we experimented with latex-based PDF reports in the past, but the issue of independent visualization of plugin results as well as the size of the included files tree have shown to be non-trivial. We will still look at this at some point but not likely in the upcoming weeks.

dorpvom, 2018-03-16:
Update: Someone is looking into pdf-report generation at the moment. I'll update this thread once real progress is made.

weidenba, 2018-05-09:
The project dorpvom mentioned has a first functional version. The project can be found [here](https://github.com/any333/FACT_analysis_report_pdf_convertor).

weidenba, 2019-03-16:
There is new [project](https://github.com/fkie-cad/fact_pdf_report) that is going to solve this issue. It will support theming so that you can easly modify the reports to suit your needs.

dorpvom, 2019-07-13:
See branch [support-beta-pdf-report](https://github.com/fkie-cad/FACT_core/tree/support-beta-pdf-report) for progess

dorpvom, 2019-08-02:
We now have a button for generating and downloading a basic pdf report. Was merged in #276.

closed 2019-08-02 by dorpvom