fkie-cad/FACT_core #86: Plugin doesn't appear under "Analysis systems"

NickDriscoll, 2018-03-21:
I followed the directions outlined [here](https://github.com/fkie-cad/FACT_core/wiki/analysis-plugin-development), yet my plugin does not show up as a checkbox on the upload page.

Are there additional steps I have to perform to add a plugin?

dorpvom, 2018-03-22:
Hi,
generally no. There should not be any additional steps.

The import of analysis plugins depends on the correct location of the file `src/plugins/analysis/<plugin_name>/code/<plugin_name>.py`, the correct name for your Plugin class, i.e. `AnalysisPlugin`. In addition, the class should inherit from `analysis.PluginBase.AnalysisBasePlugin` and implement at least the method `process_object(self, file_object)`.

If all that is the case, the plugin should be imported automatically. Could you possibly describe your setup in any detail, so that I can try to understand what might go wrong?

Johannes

NickDriscoll, 2018-03-23:
The code is in the appropriate location, the contents of the .py file look like this.
```
from analysis.PluginBase import AnalysisBasePlugin

class AnalysisPlugin(AnalysisBasePlugin):
    '''
    A plugin 
    '''
    NAME = 'name'
    DESCRIPTION = 'Description'
    DEPENDENCIES = []
    VERSION = '0.1'

    def __init__(self, plugin_administrator, config=None, recursive=True):
        '''
        recursive flag: If True recursively analyze included files
        '''
        self.config = config
 
        # additional init stuff can go here
 
        super().__init__(plugin_adminstrator, config=config, recursive=recursive, plugin_path=__file__)

    def process_object(self, file_object):
        file_object = super().process_object(file_object)

        file_object.processed_analysis[self.NAME].update({ file_object.file_path : "TEST" })

        return file_object
```

This is obviously just an example plugin.

So as you can see I do in fact create a class named AnalysisPlugin that inherits from AnalysisBasePlugin and I also implement process_object, but the plugin doesn't show up as a checkbox under "Analysis Systems"

weidenba, 2018-03-27:
Nick,

Thanks for your interest in contributing to FACT. I do not know exactly why your plug-in does not show up. But we just created a minimal functional ["hello world" analysis plug-in](https://github.com/fkie-cad/FACT_analysis_plugin_template) repository. You can just copy that and start from there. 

Peter

closed 2018-07-04 by weidenba