fkie-cad/FACT_core #94: file_path doesn't work as expected

wenter1138, 2018-04-04:
I'm trying to do some simple analysis on the paths of files in a firmware bin. I expected that the file_object.file_path string would be the path of the file as it appears in the firmware (i.e. squashfs-root/web/apache2/conf), but instead it's a giant string of hex digits plus the absolute path on my machine.

Is there a way to get what I'm looking for? If I expand the file tree after everything is analyzed, all the files and folders have the correct names, but the analysis doesn't seem to have access to that.

weidenba, 2018-04-05:
The *file_object.file_path* stores the location of the binary on your computer.  The information you are looking for is stored in *file_object.virtual_file_path*. Since a file can be present on several locations in same firmware and since the same file can be present in different firmware, the datastructure of virtual_file_path can be quite confusing. It is a dictonary of the following structure:
```python
{
    Firmware_UID: [
        'Object_UID_Level_1|Object_UID_Level_2|/actual/file/path'
    ]
}
```
 
Long answer short: The following code should provide the information, that you want:
```python
file_object.get_top_of_virtual_path(file_object.get_virtual_paths_for_one_uid()[0])
```

wenter1138, 2018-04-06:
Thanks for the response. I tried to use the statement you gave me, but it was throwing errors, so I broke it out into the following code to try and verify which statement was failing: 

```
        virtual_path_list = file_object.get_virtual_paths_for_one_uid()
        virtual_path = None
        if virtual_path_list is not None:
           virtual_path = virtual_path_list[0]
        else:
            print('get virtual paths for one uid failed!')

        file_path = None
        if virtual_path is not None:
            file_path = file_object.get_top_of_virtual_path(virtual_path)
            print('virtual path is {}'.format(virtual_path))
        else:
            print('indexing virtual path failed!')

        if file_path is None:
            file_object.processed_analysis[self.NAME]['summary'] = ['failed to acquire path']
            file_object.processed_analysis[self.NAME]['path'] = ['failed to get']
            print('get top of virtual path failed')
            return file_object
```

It worked for about the first ten minutes, showing me the file names and going through without errors. However, it started throwing "None type is not subscriptable" on 
`print('virtual path is {}'.format(virtual_path))`

Which is very strange considering I only use that print if it's not None. It's also odd that it's started doing this after a certain period of time, and hasn't stopped failing since. I've been having a lot of strange errors with statements without any kind of subscripting throwing "None type is not subscriptable". 

Anyways, do you have any insight as to why this isn't working?
Also, sorry if my formatting isn't great, I don't generally post anything on Github.

weidenba, 2018-04-09:
I just tried your code. Unfortunately I could not reproduce your problem.
Make sure that you always return the "file_object": You just return it on error in your code.
Beside that it just worked fine.
I added the complete plug-in code that worked for me below.

Maybe you ran out of memory at some point of the analysis? That might have this kind of effect. 
If that is the problem, try to reduce the "unpack _throttle_limit" in the main.cfg -> "ExpertSettings"

If that is not your problem, you may provide us your firmware sample triggering this problem. That might help us to debug FACT_core.

```python
from analysis.PluginBase import AnalysisBasePlugin

class AnalysisPlugin(AnalysisBasePlugin):
    NAME = 'support request'
    DESCRIPTION = 'support request'
    DEPENDENCIES = []
    VERSION = '0.1'

    def __init__(self, plugin_adminstrator, config=None, recursive=True):
        self.config = config

        # additional init stuff can go here

        super().__init__(plugin_adminstrator, config=config, recursive=recursive, plugin_path=__file__)

    def process_object(self, file_object):
        virtual_path_list = file_object.get_virtual_paths_for_one_uid()
        virtual_path = None
        if virtual_path_list is not None:
            virtual_path = virtual_path_list[0]
        else:
            print('get virtual paths for one uid failed!')

        file_path = None
        if virtual_path is not None:
            file_path = file_object.get_top_of_virtual_path(virtual_path)
            print('virtual path is {}'.format(virtual_path))
        else:
            print('indexing virtual path failed!')

        if file_path is None:
            file_object.processed_analysis[self.NAME]['summary'] = ['failed to acquire path']
            file_object.processed_analysis[self.NAME]['path'] = 'ERROR: failed to get the path'
            print('get top of virtual path failed')
        else:
            file_object.processed_analysis[self.NAME]['summary'] = ['path_found']
            file_object.processed_analysis[self.NAME]['path'] = file_path
        
        return file_object
```

closed 2018-10-23 by weidenba