fkie-cad/FACT_core #126: Open FACT Web server on IP Address

hbgoodman, 2018-06-25:
How do I get the FACT Webserver to run on the IP address and not the Localhost address 127.0.0.1?

dorpvom, 2018-06-26:
You can choose to install nginx along with FACT if you choose the `-N / --nginx` option with the `./install.py` script. It will initialize nginx with the config file stored in the **src/config** directory [[1]](https://github.com/fkie-cad/FACT_core/blob/master/src/config/nginx.conf).

The nginx config takes the localhost:5000 service and reverse-proxies it to :443. Now if you want to serve it on <your-ip>:443 instead of localhost:443, you can edit the nginx config accordingly.
If only an IP is intended and no domain name see [[2]](https://serverfault.com/questions/351559/nginx-without-server-name-and-using-only-static-ip-address). For specifying a domain name other than localhost see [[3]](http://nginx.org/en/docs/http/server_names.html).

Note that the fact installation with nginx will create a new self signed certificate to use for the SSL connection, which will query for some information during the process. This should obviously be replaced by a signed certificate, such as from Let's Encrypt, if used in the wild.

weidenba, 2018-06-26:
Alternatively, you might change the „http-socket“ setting in src/config/uwsgi_config.ini. If you change it to 0.0.0.0:5000, it should listen on all network interfaces.
However, if you want to deploy on your network, the nginx option might be the better choice.

Shadow0ps, 2018-07-19:
Just to echo Weidenba and Dorpvom you can do this but remember that this is not designed to be public facing and you could be opening up whatever server you have to some serious risk.  One option to help would be to configure the box with a WAF in front of it.  

Fastest way to get the box up and running on the default port (not nginx) is to change the src/config/uwsgi_config.ini file. 

`[usgwi]

# bind to the specified UNIX/TCP socket using HTTP protocol
http-socket = 123.45.67.89:5000`

Just replace "123.45.67.89" to whatever IP address your box is listening on.  If it's something like DigitalOcean you can put the public IP here but please heed the comments I mentioned about a WAF with a limited SRC ACL so you only allow access to that port (5000) from a specific IP Address or Host.

closed 2018-10-23 by weidenba