fkie-cad/FACT_core #189: Module known_vulnerabilities Bug

MasonQAQ, 2018-12-29:
Module **known_vulnerabilities** : the yara rules in src/plugins/analysis/known_vulnerabilities/signatures/vulnerabilities.yara are useless， I write a simple yara rule that can match any firmware, after run analysis, the  "known_vulnerabilities" result is still null.

weidenba, 2019-01-03:
Thanks for your feedback. We just realized that our [manual](https://github.com/fkie-cad/FACT_core/wiki/Known_Vulnerability_Rules) is missing an important step. The yara rules have to be compiled in order to be used in FACT. Normally the install script does this job.
Hopefully, running 'install.py -B' solves the problem.

We are going to write a yara compile script to support developers in the near future.

dorpvom, 2019-01-03:
Hi,
for compiling all yara signatures, including those of the _known_vulnerabilities_ plugin, there exists the **compile_yara_signatures.py** in the **src** folder. I will add a note regarding this issue to the wiki page in the next days.

dorpvom, 2019-01-04:
I've added the note [here](https://github.com/fkie-cad/FACT_core/wiki/Known_Vulnerability_Rules#use-yara-pattern-for-detection). Feel free to re-open the issue if directions are not yet clear enough.

closed 2019-01-04 by dorpvom