fkie-cad/FACT_core #194: How to remove firmware fast!

MasonQAQ, 2019-01-10:
when i delete a firmware the backend delete file for a long time ? why?

jstucke, 2019-01-10:
At the moment, the delete firmware operation is split into single delete operations which are sent over the intercom indiviually. We are aware of the problem and want to address it at some point in the future.

dorpvom, 2019-01-11:
In addition: You can also change a parameter in your config file to allow faster removal. I.e. the `ExpertSettings.intercom_poll_delay` defines the seconds between two deletions (among other things). You should be able to safely set this to say **0.1** (instead of **1.0**) to gain a speed up of 10 on the deletion time.

closed 2019-11-14 by weidenba