fkie-cad/FACT_core #209: Installation problem

gam3-over, 2019-02-07:
Hi guys ! 
i tried to install FACT_core on a super fresh machine but after i run the install.sh this is what happens:

`[2019-02-07 01:46:58][install][INFO]: Installing rampageX/firmware-mod-kit
Traceback (most recent call last):
  File "src/install.py", line 161, in <module>
    backend(distribution)
  File "/home/user/FACT_core/src/install/backend.py", line 30, in main
    _install_unpacker(distribution == 'xenial')
  File "/home/user/FACT_core/src/install/backend.py", line 128, in _install_unpacker
    'cp src/yaffs2utils/unyaffs2 src/untrx src/tpl-tool/src/tpl-tool ../../bin/'])
  File "/home/user/FACT_core/src/helperFunctions/install.py", line 150, in install_github_project
    raise InstallationError(error)
helperFunctions.install.InstallationError: Error while processing github project rampageX/firmware-mod-kit!
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... unsupported
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/sysmacros.h... no
checking for sys/stat.h... no
checking for stdlib.h... no
checking for string.h... no
checking for memory.h... no
checking for strings.h... no
checking for inttypes.h... no
checking for stdint.h... no
checking for unistd.h... no
checking for size_t... no
checking for stdlib.h... (cached) no
checking for GNU libc compatible malloc... no
checking arpa/inet.h usability... no
checking arpa/inet.h presence... yes
configure: WARNING: arpa/inet.h: present but cannot be compiled
configure: WARNING: arpa/inet.h:     check for missing prerequisite headers?
configure: WARNING: arpa/inet.h: see the Autoconf documentation
configure: WARNING: arpa/inet.h:     section "Present But Cannot Be Compiled"
configure: WARNING: arpa/inet.h: proceeding with the compiler's result
configure: WARNING:     ## ----------------------------------------------------- ##
configure: WARNING:     ## Report this to http://firmware-mod-kit.googlecode.com ##
configure: WARNING:     ## ----------------------------------------------------- ##
checking for arpa/inet.h... no
error: missing necessary header file(s)
make: *** No targets specified and no makefile found.  Stop.
`

Can you help me about that ? thank you !

dorpvom, 2019-02-07:
Hi,
PR #208 should have solved the problem. At least, it did on my local machine. Have you tried on the latest code? (PR was merged about 4 h ago)
Thanks.

gam3-over, 2019-02-07:
Hi,
first of all thank you for the reply @dorpvom 
Secondly i-ve downloaded the whole repo before than 4 hours ago.

anyway what do you mean with "running autoconf before running configure" do i have to launch some commands before setup.py ? sorry but i really can't understand

thank you very much again

dorpvom, 2019-02-07:
TL;DR Unfortunately you have to either download the most recent code version as tar-ball or clone the repository and checkout the current master. Then restart installation.

So `autconf` was previously already installed but not used for the setup of the **firmware-mod-kit** dependency. Now after a commit (to firmware-mod-kit) yesterday the setup seems to need autoconf. The referenced PR #208 solved the issue, by adding the call to autconf to the dependency setup.
So you have to use our latest code version to successfully run `install.py`.

Sorry for the inconvenience.

gam3-over, 2019-02-07:
Unfortunately i just did it but it doesnt still works.
i have downloaded it again but nothing. same error.

dorpvom, 2019-02-07:
I'll close the issue then once the #211 is merged

closed 2019-02-11 by dorpvom