fkie-cad/FACT_core #220: Unpack function timeouts

njv299, 2019-02-28:
Most of the plugins invoke external applications via the `execute_shell_command()` helper function. This function provides an optional parameter of `timeout`, which defaults to `None` (aka no timeout). It would be useful to be able to set this dynamically for each unpacker to prevent having the system choke on files that take unexpectedly long times.

I propose that each unpacker's `unpack_function()` is modified to take an additional optional `timeout` parameter (which defaults to `None`), and the unpacking scheduler can set this value via customization of the `main.cfg` file.

The same process could be done for the Analysis plugins, but they have direct access to the `config` object and therefore could each manually check for a `timeout` configuration value.

weidenba, 2019-03-16:
We are going to make some major changes regarding the backend in the near future. We will have a look if we can implement this feature during the process.

dorpvom, 2019-03-19:
This is generally possible, though it should be noted that

a) this practically means an additional (monitoring) process for each unpacker
b) the timeout can hardly be set statically since some files might take reeeaaally long (i.e. binwalk on large files) while other plugins should already time out after a couple of seconds

I guess Peter refers to the standalone extractor [[1]](https://github.com/fkie-cad/fact_extractor) [[2]](https://github.com/fkie-cad/FACT_core/tree/standalone-extractor-integration) by the way