fkie-cad/FACT_core #221: Installation - Failed to update submodules

leisipeisi, 2019-03-01:
Hi everyone, 

when trying to install FACT on Ubuntu 18.04, I am receiving the following error:
```
[2019-03-01 07:41:29][install][INFO]: FACT Installer 1.0
[2019-03-01 07:41:29][install][INFO]: Installing apt-transport-https
[2019-03-01 07:41:32][common][INFO]: Updating system
Traceback (most recent call last):
  File "src/install.py", line 154, in <module>
    common(distribution)
  File "[...]/FACT_core-master/src/install/common.py", line 37, in main
    raise InstallationError('Failed to update submodules\n{}'.format(git_output))
helperFunctions.install.InstallationError: Failed to update submodules
fatal: not a git repository (or any of the parent directories): .git
```
The line this error refers to is:
```
git_output, git_code = execute_shell_command_get_return_code('(cd ../../ && git submodule foreach "git pull")')
```
Maybe the "absolute path" is the problem?
Thanks for any help!

dorpvom, 2019-03-02:
Hi, did you download the repository as an archive?

I realize this can only work when cloning via git.
Will look at a fix ASAP.

ramesh1258, 2019-03-04:
hey i am getting error while excuting apt-get update
"Hit:1 http://old.kali.org/kali sana InRelease
                             
(appstreamcli:9629): GLib-CRITICAL **: 20:39:38.962: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Reading package lists... Done"
how to resolve this error plz help

leisipeisi, 2019-03-04:
Hey,
I indeed downloaded the repository as archive. After cloning it instead it seems to work for me, thanks for the hint!

dorpvom, 2019-03-19:
Addressed, if not really **solved** in #227