fkie-cad/FACT_core #222: (appstreamcli:9629): GLib-CRITICAL **: 20:39:38.962: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed

ramesh1258, 2019-03-04:
hey i am getting error while excuting apt-get update
"Hit:1 http://old.kali.org/kali sana InRelease

(appstreamcli:9629): GLib-CRITICAL **: 20:39:38.962: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Reading package lists... Done"
how to resolve this error plz hel

tbarabosch, 2019-03-05:
Hi, 
it seems that you're using some old Kali version. At the moment, we are not supporting Kali out of the box. Please use Ubuntu 16.04/18.04, referring to the README of FACT:
```
It is possible to install FACT on any Linux distribution but the installer is limited to Ubuntu 16.04 and 18.04 at the moment.
```

closed 2019-03-05 by tbarabosch

libingtju, 2022-07-15:
(appstreamcli:59732): GLib-CRITICAL **: 10:36:43.049: g_atomic_ref_count_dec: assertion 'old_value > 0' failed
hi，i encounter it，how to solve this problem？（Ubuntu 18.04）