fkie-cad/FACT_core #232: path variables

Celestial-intelligence, 2019-03-31:
plz make all path variables (especially when working with docker/qemu/cwe_checker) to be easy configured from .config file. when they are hardcoded as absolute paths as it is now in separate files including other repos (cwe_checker/bap) it is very hard to run framework from live boot. 

just in case everything is working great in debian jessie live boot excluding cwe_checker module and docker cuz of "no enough memory" error. 

so first i fought about changing paths in all src files, but now i see it is impossible long routine.. 
maybe you could give me an advice about linking some directories to my hard drive as just for now solution? 

i dont know exactly all the dirs that docker and all other modules uses, but still trying to figure out 

something i have already found:

git clone https://github.com/fkie-cad/cwe_checker.git internal && cd internal && docker build --build-arg=http{,s}_proxy --build-arg=HTTP{,S}_PROXY -t cwe-checker .
docker run --rm -v {}:/tmp/input {} bap /tmp/input
docker run --rm --net=none -v {}:/opt/firmware_root {} {} {}
TEST_DATA_DIR = os.path.join(get_dir_of_file(__file__), 'data/test_tmp_dir')
ENV QEMU_LD_PREFIX="/opt/firmware_root/"
FIRMWARE_ROOT=/opt/firmware_root

thank you

weidenba, 2019-04-01:
Hi,
thanks for the feedback. As far as I know the absolute paths are just present inside the docker container. Therefore it should be no problem at at all.
Furthermore, the cwe-checker docker container should be fetched from dockerhub. The clone and build feature should be triggered only, if the fetch from dockerhub fails.
Maybe there is another problem regarding your use case that I do not get at the moment.

Celestial-intelligence, 2019-04-01:
thank you for quick response
so what exactly i should do to solve low memory failure in running cwe_checker & docker from live boot?

weidenba, 2019-07-17:
closing the issue since no more feedback arrived

closed 2019-07-17 by weidenba