fkie-cad/FACT_core #237: unknown error

Celestial-intelligence, 2019-04-03:
PLEASE HELP fix this errors

[2019-04-03 21:21:21][Analysis][INFO]: Analysis Completed:
UID: 3616f441b3a44af7d2eeb26811b1d74d9445c2094a819c8c79c8720c0e699a1c_36
 Processed analysis: ['unpacker', 'file_type', 'software_components', 'binwalk', 'init_systems', 'qemu_exec', 'cpu_architecture', 'file_hashes', 'elf_analysis', 'known_vulnerabilities', 'cwe_checker']
 Files included: set()
[2019-04-03 21:21:22][PluginBase][ERROR]: Analysis worker Analysis-Worker-1 caused exception
[2019-04-03 21:21:22][PluginBase][ERROR]: Traceback (most recent call last):
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 159, in worker
    self.worker_processing_with_timeout(worker_id, next_task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 145, in worker_processing_with_timeout
    raise process.exception[0]
TypeError: 'NoneType' object is not subscriptable

[2019-04-03 21:21:22][PluginBase][ERROR]: Analysis worker Analysis-Worker-0 caused exception
[2019-04-03 21:21:22][PluginBase][ERROR]: Traceback (most recent call last):
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 159, in worker
    self.worker_processing_with_timeout(worker_id, next_task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 145, in worker_processing_with_timeout
    raise process.exception[0]
TypeError: 'NoneType' object is not subscriptable

Process ExceptionSafeProcess-157:2:
Traceback (most recent call last):
  File "/root/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 115, in _analyze_elf
    parsed_binary = lief.parse(file_object.file_path)
AttributeError: 'NoneType' object has no attribute 'parse'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/root/FACT_core/src/helperFunctions/process.py", line 59, in run
    raise exception
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 126, in process_next_object
    finished_task = self.analyze_file(task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 61, in analyze_file
    fo = self.process_object(file_object)
  File "/root/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 29, in process_object
    elf_dict, parsed_binary = self._analyze_elf(file_object)
  File "/root/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 123, in _analyze_elf
    except (TypeError, lief.bad_file) as error:
TypeError: 'NoneType' object is not subscriptable
Process Analysis-Worker-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/root/FACT_core/src/helperFunctions/process.py", line 59, in run
    raise exception
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 159, in worker
    self.worker_processing_with_timeout(worker_id, next_task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 145, in worker_processing_with_timeout
    raise process.exception[0]
[... 138 lines cut ...]
    raise exception
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 159, in worker
    self.worker_processing_with_timeout(worker_id, next_task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 145, in worker_processing_with_timeout
    raise process.exception[0]
TypeError: 'NoneType' object is not subscriptable
Process Analysis-Worker-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/root/FACT_core/src/helperFunctions/process.py", line 59, in run
    raise exception
  File "/root/FACT_core/src/helperFunctions/process.py", line 54, in run
    Process.run(self)
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 159, in worker
    self.worker_processing_with_timeout(worker_id, next_task)
  File "/root/FACT_core/src/analysis/PluginBase.py", line 145, in worker_processing_with_timeout
    raise process.exception[0]
TypeError: 'NoneType' object is not subscriptable

dorpvom, 2019-04-04:
Hi, all these errors seem to stem from the fact, that the lief module in `src/plugins/analysis/elf_analysis/code/elf_analysis.py`, which is the code of the elf_analysis plugin, is `None`.

I'm not sure how an import can be `None` without throwing an `ImportError`.
Please try to import the lief module from a python shell to check if you can import lief correctly.
If that does not work please re-run the plugin installation or just install lief: `sudo -EH pip3 install --upgrade lief`

That should fix the problem. At least in theory.

dorpvom, 2019-04-05:
Please close if that fixes your problem

Celestial-intelligence, 2019-04-17:
i have installed lief library both from source code (builded manually) and from releases, but this didn't fix the problem.
but i still can import and use lief.

attached file includes all my terminal output. FACT_core starts and works greate, but this three modules cwe_checker, elf_analyzer, qemu_exec don't work properly
[fact_core.log](https://github.com/fkie-cad/FACT_core/files/3091784/fact_core.log)

dorpvom, 2019-04-18:
It looks like the part
```
  File "/home/celestial/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 94, in create_tags
    all_funcs = self._get_relevant_imp_functions(parsed_bin.imported_functions)
  File "/home/celestial/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 73, in _get_relevant_imp_functions
    imp_functions[:] = [x for x in imp_functions if not x.startswith('__')]
  File "/home/celestial/FACT_core/src/plugins/analysis/elf_analysis/code/elf_analysis.py", line 73, in <listcomp>
    imp_functions[:] = [x for x in imp_functions if not x.startswith('__')]
AttributeError: 'lief.Function' object has no attribute 'startswith'
```
might be the root cause. If that's the case, the debugging gets easier. I will have a look on the affected code once my schedule clears up a bit.

Celestial-intelligence, 2019-04-18:
ok, thank you

dorpvom, 2019-04-20:
It seems that according to the lief docs, the values in `lief.ELF.Binary.imported_functions` should be `lief.Function` objects, like in your traceback indicated. When using lief on my machine, there are strings in that property.
The plugin, as well as the imphash helper function, rely on there being strings which causes errors in the case when there are Function objects.
This behavior seems a bit odd to me and since I cannot reproduce, its hard to come up with a workaround, other than if `isinstance ... else ...` which is kind of nasty.
Next week should see at least a short term fix for this.

dorpvom, 2019-04-21:
Fix is developed in branch [237-lief-workaround](https://github.com/fkie-cad/FACT_core/tree/237-lief-workaround).

Celestial-intelligence, 2019-04-23:
> Fix is developed in branch [237-lief-workaround](https://github.com/fkie-cad/FACT_core/tree/237-lief-workaround). PR #246.

does not fix.. :c

dorpvom, 2019-04-26:
Can you give me the output of
```
 python3 -c "import lief;print(set((type(f) for f in lief.parse('/bin/sh').imported_functions)))"
```
on your machine?

If possible, exchange `'/bin/sh'` with a binary that triggers the error in FACT.

dorpvom, 2019-05-15:
Also, version of lief and python would be interesting.

dorpvom, 2019-07-18:
I can now reproduce this. Will look at a new fix soon-ish.

dorpvom, 2019-10-02:
I'll close this since it should be fixed by #293 .
If problem persists please reopen

closed 2019-10-02 by dorpvom