fkie-cad/FACT_core #238: Login Features

mryu-org, 2019-04-09:
Hi, how can I use login to control users?

dorpvom, 2019-04-09:
Hi,
have a look at our [wiki page on authenticaion](https://github.com/fkie-cad/FACT_core/wiki/Authentication).
In short you have to enable authentication through an option in the `main.cfg` file, create a root user through the `src/manage_users.py` script and then you can use the user management tab of the WebUI (only visible when auth is enabled).

mryu-org, 2019-04-12:
> Hi,
> have a look at our [wiki page on authenticaion](https://github.com/fkie-cad/FACT_core/wiki/Authentication).
> In short you have to enable authentication through an option in the `main.cfg` file, [...]

dorpvom thanks!!!!

closed 2019-04-16 by mryu-org