fkie-cad/FACT_core #284: Add support for non standard squashfs formats

Mqr10, 2019-08-09:
sqashfs binaries which don't have a standard sqsh header are not extracted. "unsquashfs" should be added as an unpacker to help extract these non-standard binaries.

0xricksanchez, 2019-08-09:
Hi @Mqr10, 
Do you have a specific sample that is failing to extract?
I tried creating a valid squasfs via

```bash
mksquashfs . test.sqfs -noI -noD -noF -noX 
```
Afterwards I manually altered the magic sequence to `0x44444444` instead of the default of `0x73717368`. I also changed the hex value for the compression to be out of range. See [here](https://dr-emann.github.io/squashfs/).

Do you have a valid sample you could provide which makes the current squashFS fail?

Mqr10, 2019-08-12:
Hi @0xricksanchez ,

yes, I have a sample which fails. Try [https://tinyurl.com/yxppn448](url).
It does work for the first iteration step, but stops as soon as the .ubifs file is reached. Unpacking it manually with unsquashfs works without a problem. 

Thank you for your support.

0xricksanchez, 2019-08-12:
Thanks! I'll look into it within the next few days

dorpvom, 2019-08-14:
Hi, when I drop your binary into my local FACT I get this file tree:

[image]

This looks like the kernel is not extracted from the outer container (`.img`), but the inner container/fs files are handled correctly. Can you tell me on which layer you see a different result (`.img`, `.ubi`, `.ubifs`)?

Mqr10, 2019-08-14:
Hi @dorpvom,

hmm very interesting... For me it stops at the .ubifs file (see image attached).

[image]

Could this be a dependency issue? Am I missing a specific tool? The screenshot below shows the unpacker output of the .ubifs file.

[image]

Thanks a lot for your help!

dorpvom, 2019-08-14:
Okay, so on my setup, the fallback to genericFS works, i.e. mounting the file is successful.

Since the refactoring of our extraction toolchain, unpacking now runs inside a docker container that is built on docker hub and thus should be identical for all users. So a dependency issue is less likely.

If you work on a recent installation (from github) or an installation from the v3.0 tarball, I'm a bit clueless as to where this difference in behavior comes from.
Otherwise either pull the most recent version from github or just pull the latest `fact_extractor `container (`$ docker pull fkiecad/fact_extractor`) and try once more (e.g. re-do analysis).

Sorry for the inconvenience.

Mqr10, 2019-08-14:
Hi @dorpvom ,

I'm already on the latest github version and also updated the docker container following your guide. Unfortunately, I still get the same result. 
I will try to set up the system from scratch again the next days.

dorpvom, 2019-08-14:
Hi @Mqr10,

last thing: Which distro are you on?
I'm not all sure about how exactly docker sets up on top of the underlying OS kernel.
So I can't be sure that this can't also introduce different outcomes for file system operations.

Mqr10, 2019-08-14:
Hi @dorpvom,

I'm running FACT on a newly setup Ubuntu 16.04 64bit server VM.

dorpvom, 2019-08-20:
So I have set up a 16.04 VM myself. I still can not reproduce the issue though which is odd.
My libmount versions are **2.32.1** inside the container and **2.27.0** on the xenial host.
As of now I don't see a direction to look further into this :disappointed:

Mqr10, 2019-08-21:
Hi @dorpvom ,

it is definitely weird... 
I just tried to setup a VM from scratch again and still get the same error. Here are my steps to hopefully reproduce this issue on your side:

- Ubuntu Image: [http://releases.ubuntu.com/16.04/ubuntu-16.04.6-server-amd64.iso](url)
- `sudo apt update && sudo apt upgrade`
- `git clone https://github.com/fkie-cad/FACT_core.git ~/FACT_core`
- `~/FACT_core/src/install/pre_install.sh && sudo mkdir /media/data && sudo chown -R $USER /media/data`
- `sudo reboot`
- ` ~/FACT_core/src/install.py -N` (I'm installing with nginx support to access FACT from my host)
- `~/FACT_core/start_all_installed_fact_components`

weidenba, 2019-11-14:
It looks like the problem is the kernel. Depending on your virtualization system the client VM uses the host's kernel. Since the generic_fs fallback is used in this case, the kernel is responsible for extracting the filesystem. If the kernel does not support the filesystem, it fails. You can check the kernel of the vm by calling `uname -a`. If the correct kernel is used the output should include `4.15.0-45-generic #48~16.04.1-Ubuntu`.

weidenba, 2020-03-24:
It looks like this issue is solved.

closed 2020-03-24 by weidenba