fkie-cad/FACT_core #285: FACT takes very long to respond with summary

IoT-junkrat, 2019-08-15:
Hey guys,

I experience an issue, when using /rest/firmware/id?summary=true.
It took FACT 51:14 minutes to process a single request and send the 10MB JSON back.
It occurred on older versions and on the latest one.
Maybe it depends on the fact that I have several hundreds of firmware images stored in FACT?
Requests to FACT without a summary are processed in a couple of seconds..
I just need the summary of a couple of plugins and not all of them. 
Hence, would specifying the summary plugins in the url parameter help here, if one just needs a subset?

Here is the result of the verbose curl command I used.
  0 10.0M    0   99k    0     0     33      0 88:40:58  0:51:14 87:49:44
100 10.0M  100 10.0M    0     0   3427      0  0:51:14  0:51:14 --:--:--

Cheers

dorpvom, 2019-08-17:
Hi,

unfortunately this might be the expected behavior though I haven't seen such a long response time ever. Typically it takes a < 5 min on my systems.
It should be possible to develop a filter for this such that you can get summaries only for the plugins where you need them. 
Obviously you are invited to contribute on this, otherwise we'll discuss if this API change is sensible and try to have someone implement it.

Cheers

PS This run time results from some magic mongo foo that we implemented which might also a bottleneck to fix, but as of now we're not sure how / lack mongo specific expertise

IoT-junkrat, 2019-08-18:
Hey,

thanks for the answer!
Neither do I have a mongo expertise, but I just proposed the summary selection as an idea to counter the long response time. So no need to implement it, since I figured something out..

I was not sure if it is a mongo issue, a uWSGI issue or something completely else and debugged a bit. 
So what I did first, I enabled the "--thunder-lock" option in uWSGI.
This helped me to reduce the query time of the API from ~50min to ~1min.. 😀 
Maybe you guys want to add this to upstream.

Cheers

PS: Thank you guys for the invaluable work on FACT!

weidenba, 2020-03-24:
We finally added thunder-lock option by default in #374

closed 2020-03-24 by weidenba