fkie-cad/FACT_core #287: Installer Failed In Kali

sperglord8008s, 2019-08-16:
$ bash install.py
install.py: line 18: 
    FACT Installer
    Copyright (C) 2015-2018  Fraunhofer FKIE

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
: No such file or directory
No protocol specified
import-im6.q16: unable to open X server `:1' @ error/import.c/ImportImageCommand/360.
No protocol specified
import-im6.q16: unable to open X server `:1' @ error/import.c/ImportImageCommand/360.
No protocol specified
import-im6.q16: unable to open X server `:1' @ error/import.c/ImportImageCommand/360.
No protocol specified
import-im6.q16: unable to open X server `:1' @ error/import.c/ImportImageCommand/360.
from: too many arguments
install.py: line 26: try:: command not found
No protocol specified
import-im6.q16: unable to open X server `:1' @ error/import.c/ImportImageCommand/360.
from: too many arguments
from: too many arguments
from: too many arguments
from: too many arguments
from: too many arguments
from: too many arguments
install.py: line 36: except: command not found
install.py: line 37: syntax error near unexpected token `'Could not import install dependencies. Please (re-)run install/pre_install.sh''
install.py: line 37: `    sys.exit('Could not import install dependencies. Please (re-)run install/pre_install.sh')'

0xricksanchez, 2019-08-17:
Hi,

On the contrary to the `src/install/pre_install.sh`, which is is a bash script, the `src/install.py` is a python3 script. You can just invoke it as `./install.py`.

dorpvom, 2019-08-17:
Note, that the install script will quit on Kali due to this function:

```python
def check_distribution():
    codename = distro.codename().lower()
    if codename in XENIAL_CODE_NAMES:
        logging.debug('Ubuntu 16.04 detected')
        return 'xenial'
    if codename in BIONIC_CODE_NAMES:
        logging.debug('Ubuntu 18.04 detected')
        return 'bionic'
    else:
        sys.exit('Your Distribution ({} {}) is not supported. FACT Installer requires Ubuntu 16.04, Ubuntu 18.04 or compatible!'.format(distro.id(), distro.version()))
```

You can overwrite this function to say

```python
def check_distribution():
    return 'bionic'
```

but the installation process will likely fail at some point. We're looking to incorporate Kali support in the near future but as of now you would have to fix the installation problems manually.

dorpvom, 2019-08-18:
Following up on my earlier comment I tried the installation on Kali.
The things that will kill the setup are

- docker
- mongodb
- lief

These three things will not be installed correctly and thus halt the whole setup of FACT.
I've found workarounds for all three and will look to implement them as early as the upcoming week. You might want to wait for that or test own workarounds.

dorpvom, 2019-11-06:
Issues (docker, mongo, lief) parially align between debian, kali (debian based) and ubuntu >= 19.04.
docker & mongo need special mirrors while lief has to be compiled for python > 3.6.

dorpvom, 2019-11-06:
New Issue for Kali: docker-compose wants **PyYAML** in versions ` >= 3.10 < 5`. Kali comes pre installed with **PyYAML** `5.1.2` and more than a dozen packages depending on that. Meaning we can't simply remove it using apt-get. It's possible to manually prune all files installed through the `.deb` archive, but there is no telling what this might do to the other stuff depending on **PyYAML**.

PR #317 (2019-12-12): 287 support additional distros
Fixed #287 by solving installation issues for
- docker (mirrors)
- mongodb (mirrors)
- lief (dev version for py > 3.6)
- docker-compose (venv for inconsistent pyyaml version)
closed 2019-12-12 by weidenba