fkie-cad/FACT_core #316: How to run FACT_core in Pycharm IDE and debug it

040840308, 2019-11-21:
Hi, everyone

      I want to run the  FACT_core in Pycharm and debug it step by step. Does anyone know how to do this? Thanks for any advice.

dorpvom, 2019-11-21:
Hi, can you tell me which part of FACT you want to debug?
Generally you can use the `start_fact_{db, backend, frontend}.py` scripts.
For the frontend its easiest to patch the `flask_app_wrapper.py` instead of the full fledged frontend script.
When debugging the backend I'd propose starting PyCharm from a shell so that when FACT runs into an exception (and kills its process group) you don't kill your session.
Otherwise I'd always look at tests first since they allow you easier scoping of context. If you want to know where to look for I can help with that.

040840308, 2019-11-21:
> Hi, can you tell me which part of FACT you want to debug?
> Generally you can use the `start_fact_{db, backend, frontend}.py` scripts.
> For the frontend its easiest to patch the `flask_app_wrapper.py` [...]

Hi, Thanks for your reply.
I just want to debug the backend, from uploading a firmware to performing analysis. Currently, I can successfully start all the FACT related services in a shell and submit firmwares through web UI. 
I just wonder whether I can submit firmware by running in Pycharm. In this way, I can debut the backend analysis step by step. 

Thanks

dorpvom, 2019-11-21:
To test the complete analysis cycle including scheduling, blacklisting etc. you have to run/debug `start_fact_backend.py` from PyCharm (again, best to start it from a shell) and run the other two components (db, frontend) from the shell.
Debugging the complete backend, you have to keep in mind that there are mutliple layers of multiprocessing so there is no full assurance that all breakpoints you are set are triggered. But for me it works most of the time.
If you want to debug a certain plugin you can have a look at the plugin tests found in the related test folders.

040840308, 2019-11-21:
> To test the complete analysis cycle including scheduling, blacklisting etc. you have to run/debug `start_fact_backend.py` from PyCharm (again, best to start it from a shell) and run the other two [...]

OK, I see. I will have a try.
Thanks

weidenba, 2020-03-24:
@dorpvom: Should we add this to developers documentation?

dorpvom, 2020-03-25:
We could add a troubleshooting / FAQ page in the developer space with this as first entry.

dorpvom, 2020-04-22:
@040840308 Did you try the proposed method? And if so did it work?

dorpvom, 2020-12-09:
I'm going to close this for now. If interest is still there for a FAQ-style page, please re-open

closed 2020-12-09 by dorpvom