fkie-cad/FACT_core #318: no space left on device

hanjian007, 2019-11-26:
cac6fd553ef2: Verifying Checksum
cac6fd553ef2: Download complete
failed to register layer: Error processing tar file(exit status 1): write /usr/share/texlive/texmf-dist/tex/latex/ifmslide/buttonec.eps: no space left on device

cat install.log 
[2019-11-26 23:06:03][install][WARNING]: Could not update python packet pyyaml. Was not installed using pip origin

dorpvom, 2019-11-27:
The pyyaml issue should not be serious.

Can you check if you indeed have no space left on device (e.g. `$ df -h` from command line)?
FACT needs considerable storage space due to some of the used docker containers.

Best,
Johannes

hanjian007, 2019-11-27:
Filesystem                         Size  Used Avail Use% Mounted on
udev                               1.9G     0  1.9G   0% /dev
tmpfs                              393M  1.5M  391M   1% /run
/dev/mapper/ubuntu--vg-ubuntu--lv  3.9G  3.2G  515M  87% /
tmpfs                              2.0G     0  2.0G   0% /dev/shm
tmpfs                              5.0M     0  5.0M   0% /run/lock
tmpfs                              2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/loop0                          90M   90M     0 100% /snap/core/8039
/dev/loop1                          90M   90M     0 100% /snap/core/7917
/dev/sda2                          976M  145M  765M  16% /boot
tmpfs                              393M     0  393M   0% /run/user/1000

hanjian007, 2019-11-27:
My virtual machine is a 60g hard disk, isn't it enough

hanjian007, 2019-11-27:
df -h
Filesystem                         Size  Used Avail Use% Mounted on
udev                               1.9G     0  1.9G   0% /dev
tmpfs                              393M  1.5M  391M   1% /run
/dev/mapper/ubuntu--vg-ubuntu--lv  3.9G  3.2G  515M  87% /
tmpfs                              2.0G     0  2.0G   0% /dev/shm
tmpfs                              5.0M     0  5.0M   0% /run/lock
tmpfs                              2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/loop0                          90M   90M     0 100% /snap/core/8039
/dev/loop1                          90M   90M     0 100% /snap/core/7917
/dev/sda2                          976M  145M  765M  16% /boot
tmpfs                              393M     0  393M   0% /run/user/1000

dorpvom, 2019-11-27:
This line
```
Filesystem                          Size    Used    Avail   Use%    Mounted on
/dev/mapper/ubuntu--vg-ubuntu--lv   3.9G    3.2G    515M    87%     /
```
suggests your root filesystem only has 3.9 GB. If the undelying (virtual) hard disk has 60 GB - and it looks like you have LVM configured you should be able to resize the partition.

For reference I just checked a clean Debian 9 with only FACT on it. That uses 13 GB of its root partition.

closed 2019-12-03 by hanjian007