fkie-cad/FACT_core #323: Can't terminate !!!

icytxw, 2019-12-27:
When I run the command with 
```sh
 ./start_all_installed_fact_components --debug
```
the ctrl+c command can not terminate the program.

dorpvom, 2019-12-30:
Hi,
this can be due to a variety of reasons, though most trace back to a runtime error and a corresponding process not being able to be correctly terminated.

For clarification

a) Do you observe this being specifically the case with `--debug` activated?
b) Does this also occur directly (say 20 sec.) after starting the command or only after some analysis was run?
c) Can you observe any exceptions in the log?

icytxw, 2019-12-31:
@dorpvom Hi, 
Yes, the case was specifically with ```--debug``` activate, and this still occur after 24hours. Only ``` ./start_all_installed_fact_components --debug``` command trigger this issue(```./start_all_installed_fact_components and ./start_fact_backend can terminate the program normally). Cause this is not a exceptions, just can't terminate the demon process, there is no exception in the log. Can't you trigger this issue in your machine?

icytxw, 2019-12-31:
I know the reason now, the problem seems occur at function ```was_started_by_start_fac```t in  ```src/helperFunctions/program_setup.py```, which return false in function```start_fact_bactend``` when  called by ```start_all_installed_fact_components```

dorpvom, 2020-01-07:
Hi, just as an update: We'll have a look at this tomorrow. So hopefully this will be fixed soon-ish.

PR #325 (2020-01-08): Match script identifier over whole command line call to fix shutdown.…
… Bugfix credit to @icytxw.

Now wrapper scripts can be started with command line parameters and it still works.

Should close #323.
closed 2020-01-08 by weidenba