fkie-cad/FACT_core #344: Never analyzed files

IoT-junkrat, 2020-02-13:
Hey guys,

I sometimes encounter the issue that a single file analysis can not be completed (due to analysis plugin crashs or whatever reason), the UI shows - as expected- "not analyzed yet" forever.
What I am missing now is a functionality to check all files if they have been analyzed, even if the current analysis queue is showing 0 files to process
.
This feature could be run initially by starting up FACT, like an iterative command every x days or on a manual base. 

For now, even if I dig through all files in the UI file tree, I am not able to restart an analysis of a single file, because of the "Analysis might still be pending" message. So the current solution is to dig through all file trees of all binaries and re-do the analysis of the whole binary.

I hope you get my point 😀 
Would be very grateful if you consider adding this feature.

Thank you & Cheers!

dorpvom, 2020-02-13:
Hi,

just for clarification, the "not analyzed yet" and "might still be pending" messages refer to files that have not been analyzed at all yet (i.e. no plugin finished), correct?
To this point I've seen this issue only for firmware images that crash FACT somehow, e.g. by running out of memory.

If not a single plugin has run on a given file, the file must have been dropped by the scheduler somehow. Again, I've only encountered this when the scheduler itself has crashed.

Meanwhile, when all queues show 0 pending tasks, tasks may still be running on the workers. I regularly encounter this with the cwe_checker plugin. To measure this, we'd need some kind of process introspection for the worker processes. This could be achieved using a decorator. But we'd have to discuss if this is sensible in the context of our current data flow.

IoT-junkrat, 2020-02-14:
> just for clarification, the "not analyzed yet" and "might still be pending" messages refer to files that have not been analyzed at all yet (i.e. no plugin finished), correct?
Yes, that is correct.

Exactly, this happens when the problems occur, which you described.

In my case, the workers are not processing the file, since I would be "waiting" for 3 months then and I am using FACT on a powerful machine. :)

dorpvom, 2020-02-14:
So, in the meantime we've decided to test a solution for the tracking of running tasks. Once we have some action there, I'll link to this issue.
That said, your scenario ("waiting for 3 months") sounds like it should be handled by our plugin timeouts. Or is the long running time referring to the extraction part?

IoT-junkrat, 2020-02-18:
That sounds nice.
I don't have any insights what kind of mechanism broke (extraction or analysis plugin). I only see files in the tree tagged as not yet analyzed and staying there forever.
Clicking on re-do the full firmware analysis however leads to analyzing the file in a normal time. That's why I assume one of the plugins for this file crashed and never re-started the job.
This manual investigation and re-doing the analysis to hope it does not crash again does not scale and that is why I opened the ticket.

weidenba, 2020-03-31:
functionality is merged and can be accessed via the admin tab.

closed 2020-03-31 by weidenba