fkie-cad/FACT_core #359: Different database structure in CVE lookup plugin

Harald883, 2020-02-27:
When analyzing Linux kernels with the CVE lookup plugin version 0.0.3, the results are stored in MongoDB in a different structure than for other software components. For the key 'cve_results' only a string is stored instead of an object containing the name of the software and the identified CVE numbers.

It would be nice if the results of the plugin to the Linux kernel are stored in the same structure as other software components. This allows to easily map a CVE to a specific software component via MongoDB.

[image]

jstucke, 2020-02-27:
The `file_system_flag == true` indicates that the data was too large and was stored elsewhere in the file system. You can still access it trough the REST interface (see https://github.com/fkie-cad/FACT_core/wiki/rest-documentation#restfile_objectuid).
You could try something like
```sh
curl 'http://localhost:5000/rest/file_object/<UID>' -X GET
```
(you need to replace <UID> with the UID of the file object and FACT must be running)

Harald883, 2020-02-27:
Thanks for your reply! Is there a way to force FACT to store the results in the database regardless of their size?

jstucke, 2020-02-27:
Sadly no, because there is a strict size limit for documents in MongoDB. Smaller results do get stored directly in the database (`file_system_flag == false`).

dorpvom, 2020-02-27:
Hi, in addition to @jstucke s answer:

The threshold for analysis size can be configured in our config file (Default: `src/config/main.cfg`). The default value is 100 000 bytes (field `data_storage.report_threshold`).

So, while there should be some threshold here since you will get database errors for documents larger than ~ 17 MB, you can experiment with larger values. Depending on the number of plugins you run I'd guess values up to 1 MB (i.e. 1000000) should be safe.

Maybe this can already help you out.

Harald883, 2020-02-27:
Thank's a lot, I will check that out!

closed 2020-03-24 by weidenba