fkie-cad/FACT_core #365: Script update_statistic.py failes when processing too many records

Harald883, 2020-03-09:
It seems that the script 'update_statistic.py' fails when processing too many database records:

`ubuntu@db-server:~/firmware-analysis/FACT_core/src$ ./update_statistic.py
[2020-03-09 11:23:39][update_statistic][INFO]: Try to start Mongo Server...
[2020-03-09 11:23:39][MongoMgr][INFO]: start local mongo database
Traceback (most recent call last):
  File "./update_statistic.py", line 50, in <module>
    sys.exit(main())
  File "./update_statistic.py", line 39, in main
    updater.update_all_stats()
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 42, in update_all_stats
    self.db.update_statistic('unpacking', self.get_unpacking_stats())
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 247, in get_unpacking_stats
    self.db.file_objects, '$processed_analysis.unpacker.entropy', unwind=True, match={'processed_analysis.unpacker.summary': 'unpacked'})).keys())
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 376, in _get_objects_and_count_of_occurrence_single_db
    tmp = get_objects_and_count_of_occurrence(database, object_path, unwind=unwind, match=merge_dict(match, self.match))
  File "/usr/local/lib/python3.6/dist-packages/common_helper_mongo/aggregate.py", line 84, in get_objects_and_count_of_occurrence
    result = list(collection.aggregate(pipeline))
  File "/usr/local/lib/python3.6/dist-packages/pymongo/collection.py", line 2380, in aggregate
    **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/collection.py", line 2299, in _aggregate
    retryable=not cmd._performs_write)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/mongo_client.py", line 1464, in _retryable_read
    return func(session, server, sock_info, slave_ok)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/aggregation.py", line 148, in get_cursor
    user_fields=self._user_fields)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/pool.py", line 613, in command
    user_fields=user_fields)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/network.py", line 167, in command
    parse_write_concern_error=parse_write_concern_error)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/helpers.py", line 159, in _check_command_response
    raise OperationFailure(msg % errmsg, code, response)
pymongo.errors.OperationFailure: Sort exceeded memory limit of 104857600 bytes, but did not opt in to external sorting. Aborting operation. Pass allowDiskUse:true to opt in.`

According to this article the 'allowDiskUse' parameter in aggregate commands could solve the problem:
https://mkyong.com/mongodb/mongodb-sort-exceeded-memory-limit-of-104857600-bytes/

weidenba, 2020-03-10:
Thanks for the note. Just for the record, how many files do we need in our database to reproduce this issue? We are going to have a look at this next week.

Harald883, 2020-03-10:
This issue occures in the method `get_unpacking_stats`. Last time I was able to run the script successfully, I had about 560K unique unpacked files. So I guess it should be about 600K to 650K files now.

jstucke, 2020-03-18:
#368 hopefully fixes the problem. If you like, you can check out the branch and give it a try.

Harald883, 2020-03-19:
Unfortunately, this does not seem to fix the bug. The script fails with exactly the same error message as before:

```
[2020-03-19 09:59:13][update_statistic][INFO]: Try to start Mongo Server...
[2020-03-19 09:59:13][MongoMgr][INFO]: start local mongo database
Traceback (most recent call last):
  File "./update_statistic.py", line 50, in <module>
    sys.exit(main())
  File "./update_statistic.py", line 39, in main
    updater.update_all_stats()
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 42, in update_all_stats
    self.db.update_statistic('unpacking', self.get_unpacking_stats())
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 243, in get_unpacking_stats
    self.db.file_objects, '$processed_analysis.unpacker.entropy', unwind=True, match={'processed_analysis.unpacker.summary': 'unpacked'})).keys())
  File "/home/ubuntu/firmware-analysis/FACT_core/src/statistic/update.py", line 373, in _get_objects_and_count_of_occurrence_single_db
    tmp = get_objects_and_count_of_occurrence(database, object_path, unwind=unwind, match=merge_dict(match, self.match))
  File "/usr/local/lib/python3.6/dist-packages/common_helper_mongo/aggregate.py", line 84, in get_objects_and_count_of_occurrence
    result = list(collection.aggregate(pipeline))
  File "/usr/local/lib/python3.6/dist-packages/pymongo/collection.py", line 2380, in aggregate
    **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/collection.py", line 2299, in _aggregate
    retryable=not cmd._performs_write)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/mongo_client.py", line 1464, in _retryable_read
    return func(session, server, sock_info, slave_ok)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/aggregation.py", line 148, in get_cursor
    user_fields=self._user_fields)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/pool.py", line 613, in command
    user_fields=user_fields)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/network.py", line 167, in command
    parse_write_concern_error=parse_write_concern_error)
  File "/usr/local/lib/python3.6/dist-packages/pymongo/helpers.py", line 159, in _check_command_response
    raise OperationFailure(msg % errmsg, code, response)
pymongo.errors.OperationFailure: Sort exceeded memory limit of 104857600 bytes, but did not opt in to external sorting. Aborting operation. Pass allowDiskUse:true to opt in.
```

dorpvom, 2020-03-19:
Indeed looking at the final non pymongo line of the traceback

```
File "/usr/local/lib/python3.6/dist-packages/common_helper_mongo/aggregate.py", line 84, in get_objects_and_count_of_occurrence
    result = list(collection.aggregate(pipeline))
```

we can see that the error is not coming from FACT code but from the `common_helper_mongo` library (which is maintained by us as well). I'll spin up a bugfix there.

dorpvom, 2020-03-19:
Okay, bug fix is merged in library. You can update the dependency by running:

```sh
sudo -EH pip3 install -U git+https://github.com/fkie-cad/common_helper_mongo.git
```

Harald883, 2020-03-19:
Thanks a lot, this in combination with [#368](https://github.com/fkie-cad/FACT_core/pull/368) fixed the bug!

weidenba, 2020-03-24:
#368 is going to merged soon. Therefore, I close this issue.

closed 2020-03-24 by weidenba