fkie-cad/FACT_core #369: Installation Failure

stephen3303, 2020-03-18:
VMWare Ubuntu 16.04
run all steps in install.md up to install.py resulting in the following output.

[2020-03-18 09:04:36][install][INFO]: FACT Installer 1.0
[2020-03-18 09:04:36][install][INFO]: Installing apt-transport-https
[2020-03-18 09:04:36][common][INFO]: Updating system
[2020-03-18 09:04:38][install][INFO]: Installing python3 python3-dev build-essential automake autoconf libtool git unzip
[2020-03-18 09:04:38][install][INFO]: Removing python3-pip python3-setuptools python3-wheel
[2020-03-18 09:04:39][common][INFO]: Installing python3 pip
[2020-03-18 09:04:42][install][INFO]: Installing python python-dev
[2020-03-18 09:04:43][install][INFO]: Removing python-pip
[2020-03-18 09:04:43][common][INFO]: Installing python2 pip
[2020-03-18 09:04:45][install][INFO]: Installing libmagic-dev
[2020-03-18 09:04:46][install][INFO]: Installing libfuzzy-dev
[2020-03-18 09:04:46][install][INFO]: Installing python3-tlsh
[2020-03-18 09:04:47][install][INFO]: Installing git+https://github.com/fkie-cad/fact_helper_file.git
[2020-03-18 09:04:49][install][INFO]: Installing psutil
[2020-03-18 09:04:50][install][INFO]: Installing pytest==3.5.1 pytest-cov pytest-flake8 pylint python-magic xmltodict yara-python==3.7.0 appdirs
[2020-03-18 09:04:57][install][INFO]: Installing ssdeep
[2020-03-18 09:04:58][install][INFO]: Installing lief
Traceback (most recent call last):
  File "src/install.py", line 173, in <module>
    install()
  File "src/install.py", line 154, in install
    common(distribution)
  File "/home/ubuntu/FACT_core/src/install/common.py", line 64, in main
    pip3_install_packages('lief')
  File "/home/ubuntu/FACT_core/src/helperFunctions/install.py", line 95, in pip3_install_packages
    return _pip_install_packages(3, args)
  File "/home/ubuntu/FACT_core/src/helperFunctions/install.py", line 78, in _pip_install_packages
    raise installation_error
  File "/home/ubuntu/FACT_core/src/helperFunctions/install.py", line 73, in _pip_install_packages
    run_shell_command_raise_on_return_code('sudo -EH pip{} install --upgrade {}'.format(version, packet), 'Error in installation of python package {}'.format(packet), True)
  File "/home/ubuntu/FACT_core/src/helperFunctions/install.py", line 51, in run_shell_command_raise_on_return_code
    raise InstallationError(error)
helperFunctions.install.InstallationError: Error in installation of python package lief
Collecting lief
  Using cached lief-0.10.1.tar.gz (12.7 MB)
Building wheels for collected packages: lief
  Building wheel for lief (setup.py): started
  Building wheel for lief (setup.py): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-yje_i44d/lief/setup.py'"'"'; __file__='"'"'/tmp/pip-install-yje_i44d/lief/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-112lwu6o
       cwd: /tmp/pip-install-yje_i44d/lief/
  Complete output (46 lines):
  0.10.1
  running bdist_wheel
  running build
  running build_ext
  Traceback (most recent call last):
    File "/tmp/pip-install-yje_i44d/lief/setup.py", line 72, in run
      subprocess.check_output(['cmake', '--version'])
    File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
      **kwargs).stdout
    File "/usr/lib/python3.5/subprocess.py", line 693, in run
      with Popen(*popenargs, **kwargs) as process:
    File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
      restore_signals, start_new_session)
    File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
      raise child_exception_type(errno_num, err_msg)
  FileNotFoundError: [Errno 2] No such file or directory: 'cmake'
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-yje_i44d/lief/setup.py", line 367, in <module>
      version=version
    File "/usr/local/lib/python3.5/dist-packages/setuptools/__init__.py", line 144, in setup
      return distutils.core.setup(**attrs)
[... 57 lines cut ...]
      File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/local/lib/python3.5/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.5/distutils/command/install.py", line 583, in run
        self.run_command('build')
      File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.5/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-yje_i44d/lief/setup.py", line 75, in run
        ", ".join(e.name for e in self.extensions))
    RuntimeError: CMake must be installed to build the following extensions: lief
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-yje_i44d/lief/setup.py'"'"'; __file__='"'"'/tmp/pip-install-yje_i44d/lief/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-z3xbf537/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.5/lief Check the logs for full command output.

weidenba, 2020-03-19:
Thanks for the note. Do you use ubuntu 16.04 server or desktop version?

dorpvom, 2020-03-19:
It seems like cmake is missing during setup. We don't have it in our installation. You can try install it by hand and see if setup runs successfully afterwards. If so, we can add cmake as explicit dependency in our setup.

stephen3303, 2020-03-19:
Desktop

stephen3303, 2020-03-20:
pip install cmake
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cmake
  Downloading https://files.pythonhosted.org/packages/a5/7c/6525cadf99abbabbcb29676f53de0441e8d2f8d0114ab52aae2b31223a3b/cmake-3.16.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-cJXVXE/cmake/setup.py", line 7, in <module>
        from skbuild import setup
    ImportError: No module named skbuild
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-cJXVXE/cmake/

weidenba, 2020-03-20:
I tried to reproduce your issue on a newly installed ubuntu 16.04.6 Desktop system. Unfortunately I was not able to trigger this issue. However you can try to install build-essential and cmake via apt-get.  
`sudo apt-get install build-essential cmake`

Another issue could be an old version of setuptools. So you may try:
`sudo -EH pip3 install setuptools --upgrade`

aadithya96, 2020-06-08:
Reproduced same issue in Ubuntu 20.04 Server edition. 
Tried to resolve by 'sudo apt install cmake', but that seemed to cause issues with lief. 
Will post updates and logs.

dorpvom, 2020-06-09:
Hi,

yes logs would be helpful.

I'm still not sure why this even happens. Also, we have a 20.04 CI build that typically builds successfully (though we have some curious issues popping up when running our integration tests ...). So at least it should be able to install fully on 20.04.

Anyway, we'll look into it some more.

dorpvom, 2020-12-09:
Since this is a bit dated by now, and we have changed a number of installation things in the meantime, I'll close this for now. Please post some logs if you re-open

closed 2020-12-09 by dorpvom