fkie-cad/FACT_core #375: Stored XSS when updating analysis details

lum8rjack, 2020-03-29:
The application does not sanitize the user input when updating the details of a firmware. When updating the firmware details, the applications posts the data to the "upload-analysis/[uid]" page. That page then runs the function "_get_meta_from_request()" within the /src/helperFunctions/mongo_task_conversion.py script. I tested the "tags" and "version" fields by adding JavaScript to the updated fields (it looks like other fields are possibly vulnerable too). This data is not sanitized and is stored in the application. Each time a page is requested with this information, the JavaScript is ran. I added some screenshots to help. Using the "escape" function within flask could be a solution to sanitize the input. I tested it against the version field and it worked. Let me know if there are any additional questions.

[image]
Adding js to the tags field
[image]
When browsing to the home page it loads the js
[image]
js is executed
[image]
Used the escape function to sanitize the input
[image]

dorpvom, 2020-03-30:
Hi,

we actually did not do a great job securing our web application, thus the warning in our *readme*. That said we're happy to improve security wherever possible. Mostly we simply lack the background in web application design.
Would you like to open a Pull Request on that yourself, since you seem to have a solution already?

Best,
Johannes

lum8rjack, 2020-03-30:
Thanks for the quick response. Yeah I can test the solution to make sure it works and doesn't affect anything else and then open a pull request.

Clint

weidenba, 2020-03-31:
fix merged

closed 2020-03-31 by weidenba