fkie-cad/FACT_core #395: Make more cross-platform.

kallisti5, 2020-04-19:
Why is Ubuntu / Debian so embedded into the code for FACT?

Logic around "setting up docker" and "installing basic buildtools" is embedded within the python code.

This guarantees:

1) FACT can never be packaged on anything except a few versions of ubuntu unless knowledge of newer versions is added to the python code. (aka, this codebase will not age well)
2) FACT can never be ported to other distributions such as Fedora/Centos/etc without major rework of the python code.

Some thoughts:
* The pre-install.py could do a lot of the "os prep" stuff done in "src/install/common.py"
* This would enable /src/install/* to focus on non-os setup tasks.

Distro packagers then could bypass the pre-install.py step, and handle that via packaging system manifests.

kallisti5, 2020-04-19:
Here's what happens when you try to pre-install, and run the FACT from the source directory:

```
$ sudo ./src/start_fact.py -d
[2020-04-18 22:06:55][start_fact][DEBUG]: db not installed
[2020-04-18 22:06:57][start_fact][DEBUG]: frontend not installed
[2020-04-18 22:06:57][start_fact][DEBUG]: backend not installed
Traceback (most recent call last):
  File "./src/start_fact.py", line 101, in <module>
    if not _process_is_running(backend_process):
  File "./src/start_fact.py", line 81, in _process_is_running
    os.kill(process.pid, 0)
AttributeError: 'NoneType' object has no attribute 'pid'
```

Rummaging through the code, it appears the app expects a config directory in the working path populated with things configured / deployed via the python install scripts.

So yeah, drop the "os setup" stuff from the src/install.py, move it over to pre_install.sh and FACT might be able to run on platforms other than Debian-based distros.

dorpvom, 2020-04-20:
Hi,

we have talked about more platform independence for a while. When you look at our commit history you'll see that we actually only support debian and kali for under a year. So that alone took a bit of time to get to.

That said, your proposal, while neat, lacks in one aspect: We currently have structured the installation such that frontend (ui),  backend (analysis), and database can be set up independently and with only a minimal amount of unnecessary packets installed across all machines. This would have to be kept in mind in a new installation procedure.

The most amount of work though, in my view is finding the packages for rpm, yum and so on that are equal to the apt packages that we depend on. If you like to contribute in this aspect regarding rpm packages this would go a long way towards support for rpm based distributions.

We currently have no one working on better platform support, but I'll try to find someone to look into this. Maybe we find a way to both offer an easier way of adding platforms and keep the footprint on independent components small.

Best regards and thanks for all contributions
Johannes

kallisti5, 2020-04-20:
It appears all dependencies run in containers... so why not put FACT into a container?  Then it wouldn't matter what OS anyone is running.

dorpvom, 2020-04-21:
Possible. Something we have been discussing. Or more specifically having a set of interconnected containers, each for a separate FACT part. But that requires some engineering regarding the data flow.
The "one container" solution will probably not work out of the box for nesting reasons. For now we have defaulted to VMs for that kind of solution.

Note that we do want a more flexible setup. It just has not been emphasized yet, since we had no outside reason for it.

dorpvom, 2020-06-17:
@any333 is working on this at the moment right [here](https://github.com/any333/FACT_core/tree/fedora)

any333, 2020-07-08:
The new installation was merged into master last week, so it is possible now to run FACT under Fedora.

closed 2020-07-15 by any333