fkie-cad/FACT_core #397: ip_and_uri_finder start failed

cc-crack, 2020-04-20:
ip_and_uri_finder start failed while executing  start_all_installed_fact_components.

Here is the boot log.
```
2020-04-20 16:59:35][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.6/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2020-04-20 16:59:35][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.6/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
Traceback (most recent call last):
  File "/xxx/FACT_core-3.0/src/../start_fact_backend", line 52, in <module>
    analysis_service = AnalysisScheduler(config=config)
  File "/xxx/FACT_core-3.0/src/scheduler/Analysis.py", line 32, in __init__
    self.load_plugins()
  File "/xxx/FACT_core-3.0/src/scheduler/Analysis.py", line 171, in load_plugins
    plugin.AnalysisPlugin(self, config=self.config)
  File "/xxx/FACT_core-3.0/src/plugins/analysis/ip_and_uri_finder/code/ip_and_uri_finder.py", line 44, in __init__
    self.reader = geoip2.database.Reader(str(GEOIP_DATABASE_PATH))
  File "/usr/local/lib/python3.6/dist-packages/geoip2/database.py", line 85, in __init__
    self._db_reader = maxminddb.open_database(fileish, mode)
  File "/usr/local/lib/python3.6/dist-packages/maxminddb/__init__.py", line 41, in open_database
    return maxminddb.reader.Reader(database, mode)
  File "/usr/local/lib/python3.6/dist-packages/maxminddb/reader.py", line 51, in __init__
    with open(database, 'rb') as db_file:
FileNotFoundError: [Errno 2] No such file or directory: '/xxx/FACT_core-3.0/src/plugins/analysis/ip_and_uri_finder/bin/GeoLite2-City/GeoLite2-City.mmdb'
```
It seems that GeoLite2-City.mmdb download failed while this plugin installation.

```
wget http://geolite.maxmind.com/download/geoip/database/GeoLite2-City.tar.gz
--2020-04-20 17:03:45--  http://geolite.maxmind.com/download/geoip/database/GeoLite2-City.tar.gz
Resolving geolite.maxmind.com (geolite.maxmind.com)... failed: Name or service not known.
wget: unable to resolve host address ‘geolite.maxmind.com’
```

```
nslookup  geolite.maxmind.com
Server:         127.0.0.53
Address:        127.0.0.53#53

** server can't find geolite.maxmind.com: NXDOMAIN
```

cc-crack, 2020-04-20:
https://blog.maxmind.com/2019/12/18/significant-changes-to-accessing-and-using-geolite2-databases/

Maxmind required a user license key to download GeoLite2 databases.

jstucke, 2020-04-20:
Sadly, we can't use the GeoLite database anymore because of those licensing issues.
I'm pretty sure we fixed the error at some point, though. Are you using the latest version of FACT?

cc-crack, 2020-04-20:
Well, I'm using the latest official release v3.0 released on Jul 31, 2019, which doesn't have your fix yet. Thanks. Someone mirrored the [database](https://github.com/DocSpring/geolite2-city-mirror). I've solved this issue.

closed 2020-04-20 by cc-crack