fkie-cad/FACT_core #400: Know if a firmware analysis has been completed

MrSuicideParrot, 2020-04-20:
I'm developing an application that uses the FACT REST API and I would like to know what is the best way to find out if all tasks related to an analysis have been completed. 

When I was reading the documentation, I didn't see any way to retrieve this information directly.

weidenba, 2020-04-21:
At the moment there is no easy way to check if a specific firmware analysis is finsihed. However you can check, if there is something left in the analysis queues. 
The `/rest/status` [endpoint](https://github.com/fkie-cad/FACT_core/wiki/rest-documentation#reststatus) provides a json document including this information `system_status -> backend -> unpacking | analysis`.

I think @dorpvom figured out a way to check if results of included files are still missing. I am going to ask him if he can share his knowledge with us.

dorpvom, 2020-04-21:
My solution was to recursively check a firmware, extraction layer by extraction layer, for if one of the extracted files is missing in the database. This can either be done using the database directly, or using the REST API. This is not a high performance solution, but it works.

For load balancing I use the solution presented by @weidenba regarding the /status REST endpoint.

The issue btw. is that we neither track an analysis id of some sort, nor have a comprehensive introspection into our queues.

dorpvom, 2020-04-29:
#407 brings a feature to track the analysis progress of specific firmware images

dorpvom, 2020-04-30:
I'll close this for now since #407 was merged. If you have a question on how to get the information using the UI or the REST API, reopen or rename issue.

closed 2020-04-30 by dorpvom

jstucke, 2020-04-30:
#407 is merged into master now. You should be able to see currently running analyses on the system_health page. What is more, you can also access the information via REST:
```bash
curl "http://localhost:5000/rest/status" -X GET | python3 -m json.tool
```

```json
"system_status": {
        "backend": {
            "_id": "backend",
            "analysis": {
                "analysis_main_scheduler": 0,
                "current_analyses": {
                    "7ed17735d32d9d0cae78fd8b48f9edfc43e9419f6b9b49abe0044a172c13e170_14274560": 10
                },
```

jstucke, 2020-04-30:
For clarification: the keys in `current_analyses` are firmware UIDs and the value is the number of files unpacked from the firmware which are not fully analyzed yet. This number can increase when more included files get unpacked recursively. When the analysis is complete, the UID entry should disappear from the json object/dict.

MrSuicideParrot, 2020-04-30:
Thanks so much for your help!  I will update my code to use this feature.