fkie-cad/FACT_core #430: binwalk can't find file

weina2016, 2020-06-18:
fact 3.1 when I analysised a firmware, I got this error.
Could not read file: FileNotFoundError [Errno 2] No such file or directory: '/tmp/fact_analysis_binwalk_i1wt7ay2/0014a5091b2f27e25c1c897e399eb1c3e6f455e7059f358f46f2c26203729665_5135728.png'

dorpvom, 2020-06-18:
Hi @weina2016  ,
can you provide additional context for the issue including more of the traceback?
It seems that the entropy graph image is missing but for more take aways I need more context.

dorpvom, 2020-06-18:
Also if the file `0014a5091b2f27e25c1c897e399eb1c3e6f455e7059f358f46f2c26203729665_5135728` is not sensitive you might provide a link so I can try to reproduce the issue

weina2016, 2020-06-19:
Hi @dorpvom , 
Thank you for your response. I just got this error in the terminal .

[2020-06-19 09:45:34][fail_safe_file_operations][ERROR]: Could not read file: FileNotFoundError [Errno 2] No such file or directory: '/tmp/fact_analysis_binwalk_xzoqqqis/9f3a4aecc9950d150eee999fce6597af74a50e11727655e994541186cdfd9caa_14443038.png'

No more detail.And I tried other firmwares , got  errors like this. Did I miss any docker images?

weina2016, 2020-06-19:
The command "xvfb-run -a binwalk -BEJ " didn't generate any files

dorpvom, 2020-06-19:
Have you tried running binwalk without frame buffer (i.e. `$ binwalk -BEJ `)? Binwalk should be installed from sources, so it should be consistent accross platforms. Maybe you have an issue with xvfb or this is another `/tmp` folder issue (see #427).

dorpvom, 2020-12-09:
Hi. Is this still an issue? I'd like to close it otherwise.

rhelmke, 2021-11-12:
It has been a while 🤔. Closing for now.

closed 2021-11-12 by rhelmke