fkie-cad/FACT_core #436: Remote Host Configuration - Radare View redirects to localhost:8000

nstarke, 2020-06-26:
Hi,
I am using FACT on a remote host using the nginx configuration.  When I select `view in radare` I am redirected to `localhost:8000`, which of course doesn't work with a remote host. 

I'm not sure of the best way to approach solving this problem, but would be happy to submit a pull request if you can guide me on how you would like to fix this issue.  Thanks for your help.

dorpvom, 2020-06-26:
Hi,

I know that this is a bit of a hassle, but I did not have a perfect idea how to handle this generically back when it was implemented.
I documented the necessary steps [here](https://github.com/fkie-cad/FACT_core/wiki/radare-integration) (see the _ssl and hostname issues_ paragraph).
Basically the IP address of your host had to be added somewhere. A possible way of handling this though only the `main.cfg` would be to store the `localhost` string in the config, read it in the helper function and for a _remote_ setup replace localhost with your IP in the config instead of in the code.

I'm pretty sure the described steps work since we have it running for some time now and I configured one such setup rather recently. But I'd be curious to hear if / how it works for you.

Best
Johannes

nstarke, 2020-06-26:
Thank you for your response, I will reconfigure one of my FACT clusters next week to test changing the URL in code, then I will try to make this configuration driven.  

Is there anyway using nginx to proxy the traffic for a specific route to the radare web service?  That way everything could run over 443 along with the rest of the application.  Just a thought.

nstarke, 2020-06-26:
I see now that by setting `nginx` in `ExpertSettings` you can do what I was asking above.  Sorry to ask a question that has already been answered.  Pull request forthcoming.

PR #437 (2020-06-30): Making radare web service host/ip configurable
This PR makes the radare2 host url configurable via the main.cfg
file. The default value is set to 'localhost', but should be set
to the IP or hostname of where ever the radare service is running
in distributed configurations.

Fixes #436
closed 2020-06-30 by jstucke