fkie-cad/FACT_core #439: REST API does not return results of software_components analysis

bauerj, 2020-06-29:
If I use the REST API to get information about an image, most analysis have an empty summary:

```
johann@firmware:~/tool$ curl http://localhost:5000/rest/firmware/dda26c1b592153a178665b45d65a7817c22f22d545b26fa3573ce0b31b72d483_38793190?limit=1000 | python3 -m json.tool
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 29471  100 29471    0     0  2398k      0 --:--:-- --:--:-- --:--:-- 2398k
{
    "firmware": {
        "analysis": {
            "binwalk": {
                "analysis_date": 1592482729.966519,
                "entropy_analysis_graph": "[redacted]",
                "plugin_version": "0.5.2",
                "signature_analysis": "\nDECIMAL       HEXADECIMAL     DESCRIPTION\n--------------------------------------------------------------------------------\n0             0x0             gzip compressed data, from Unix, last modified: 2020-05-26 10:33:44\n\n",
                "summary": [
                    "gzip compressed data"
                ]
            },
            "cpu_architecture": {
                "analysis_date": 1592482737.8192575,
                "plugin_version": "0.3.2",
                "summary": []
            },
            "crypto_hints": {
                "analysis_date": 1592482712.1785395,
                "plugin_version": "0.1",
                "summary": [],
                "system_version": "3.7.1_1592482075"
            },
            "crypto_material": {
                "analysis_date": 1592482713.1150856,
                "plugin_version": "0.5.2",
                "summary": [],
                "system_version": "3.7.1_1592482075"
            },
            "cve_lookup": {
                "analysis_date": 1592482732.2040658,
                "cve_results": {},
                "plugin_version": "0.0.4",
                "summary": []
            },
            "elf_analysis": {
                "analysis_date": 1592482737.5556078,
                "plugin_version": "0.3",
                "skipped": "blacklisted file type",
                "summary": []
            },
            "file_hashes": {
                "analysis_date": 1592482736.9007568,
                "imphash": null,
                "md5": "a9116bb9ad7537b53dbec2cfbc5009df",
                "plugin_version": "1.1",
                "ripemd160": "68f89bcc4ce1023057793df451eef06c37cf572b",
                "sha1": "7f31bacb1abcb2d39d94af3199f0cc269bb36025",
                "sha256": "dda26c1b592153a178665b45d65a7817c22f22d545b26fa3573ce0b31b72d483",
                "sha512": "c5c21b3d8287dcc08dfd4663607d3236e3d5598604cf89b30ed6f5317e2030fc9d508ef22c9d065f35f025d41e73b0a7c2fa333a3966c16efa794d62054aa3d2",
                "ssdeep": "786432:usfICUZqbA02FTd7FZhdPPYday4gbome/7oalw:usQ9kkl9dBZh63lUmezlw",
                "tlsh": "0987332EA0B9822CEE7C76341F723D0FCE4406CA16A4DEB5A1FD5F30B2AF512156D991",
                "whirlpool": "e61e69c7417d3bd107145d54e6e5212100b65ff74cd3c04eece96fcc5cf66bdd3d6a2430e4dc7af6495a411d9c8b2f8033cafcde577e772d23893baa52dd3631"
            },
            "file_type": {
                "analysis_date": 1592482688.0426052,
                "full": "gzip compressed data, last modified: Tue May 26 10:33:44 2020, from Unix, original size modulo 2^32 48179200",
                "mime": "application/gzip",
                "plugin_version": "1.0",
                "summary": [
                    "application/gzip"
                ]
            },
[... 1235 lines cut ...]

                , Type: application/x-executable

        </span>
        <a href="location.href='/download/101deea080642303a91a6898c94ac52255fd9551da9dd6408a25ff42a155f8c6_1582488'"><i class="fas fa-download"></i></a>
    </p>

            </span>
            <span class="btn-group">
                <span class="btn btn-sm btn-primary"><i class="fas fa-list-ul"></i></span>
                <span class="btn btn-sm btn-primary">0</span>
            </span>
        </div>
        <div id="kCwHNCgciQ" class="collapse">

        </div>

</div>
                </td>
            </tr>

</table>
```

Is this an API bug or do I have to use a different request to get this data in a more structured form?

dorpvom, 2020-06-29:
The API by default does not compute the summaries, since that is quite a heavy computation and response time suffers. You have to enforce this with the `summary` parameter (see API docu in wiki for reference).

`$ curl http://localhost:5000/rest/firmware/dda26c1b592153a178665b45d65a7817c22f22d545b26fa3573ce0b31b72d483_38793190?summary=true`
(Note: the limit parameter does not have an effect on single firmware request)

dorpvom, 2020-06-29:
I'd encourage our [gitter](https://gitter.im/FACT_core/community) channel for general questions. But as long as we roam around 15 -20 issues the issue way is manageable.