fkie-cad/FACT_core #441: Webhook for finished firmware analysis

nstarke, 2020-06-30:
This is an idea that I think would be useful.

It would be great if there was a "webhook" that would make an HTTP request to a specified server when a firmware image analysis is complete.  The server address could either be specified during the upload process, or configured as part of main.cfg.  This would greatly help automated importing.  Right now my automation scripts dump everything into FACT and quickly overload the server.  A webhook would allow for measured uploading.  

Just an idea, you can feel free to close this ticket if you disagree.  If you like the idea, I'm happy to work on the implementation.

jstucke, 2020-06-30:
Hi, I think this is a great idea. I already thought about whether e-mail notification would be a useful addition but this would probably be even better. Your contribution will be very much appreciated!

jstucke, 2020-06-30:
Since FACT is designed to be a multi-user system, making the feature configurable during upload would make the most sense in my opinion.

nstarke, 2020-07-19:
I've been looking at the FACT code to try to discern the best place to insert code for this feature, and I'm a bit at a loss.  Where would be the preferred location for this code?  I'm looking specifically for when a workload is complete.  Thanks for your help!

jstucke, 2020-07-20:
The function `result_collector()` in `src/scheduler/Analysis.py` is used for collecting the results of the analysis plugins. In `_remove_from_current_analyses()` in line 473 the check, whether the firmware analysis is complete takes place. This is probably place where the hook should be triggered.

nstarke, 2020-07-20:
I tried placing the code at line 483 in that file but the `fw_object` object was always of type `FileObject` at the end of the analysis and thus I couldn't access the `webhook_url` property I added to the `Firmware` object.  I agree with you that that looks like the proper location, but I would need some help figuring out how to get the webhook_url into that function at that time.

jstucke, 2020-07-21:
The `Firmware` object and any unpacked `FileObject` undergo the analysis process individually. This made the check, if the analysis is complete a bit complicated. But you could simply add the webhook url to the data in `_init_current_analysis()` when the `Firmware` object is analyzed and retrieve it in `_remove_from_current_analyses()` when you know the analysis is finished (when the analysis of the last recursively unpacked `FileObject` is done).

jstucke, 2020-07-23:
I was thinking about this feature and I noticed another problem: the analysis scheduler (where the completed analysis is noticed) and the frontend (from where the signal is sent) may be running on different systems if FACT was set up as a distributed system. Therefore, the information needs to run through the "intercom" (which sits in between). This would require some new logic there is well.

jstucke, 2024-11-13:
I will close this issue since it has been stale for a while. Feel free to re-open it if this is still something you're interested in.

closed 2024-11-13 by jstucke