fkie-cad/FACT_core #442: Error: MongoDB can only handle up to 8-byte ints

nstarke, 2020-06-30:
I'm receiving this error message when running analysis on a firmware image:

```
[2020-06-30 15:44:03][db_interface_backend][ERROR]: Update of analysis failed badly (MongoDB can only handle up to 8-byte ints)                         Process ExceptionSafeProcess-51:
Traceback (most recent call last):                                                                                                                        File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/nick/FACT_core/src/helperFunctions/process.py", line 57, in run                                                                               raise exception
  File "/home/nick/FACT_core/src/helperFunctions/process.py", line 52, in run                                                                               Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run                                                                                    self._target(*self._args, **self._kwargs)
  File "/home/nick/FACT_core/src/scheduler/Analysis.py", line 378, in result_collector                                                                      self.post_analysis(fw)
  File "/home/nick/FACT_core/src/storage/db_interface_backend.py", line 178, in add_analysis                                                                self._update_analysis(file_object, analysis_system, processed_analysis[analysis_system])
  File "/home/nick/FACT_core/src/storage/db_interface_backend.py", line 197, in _update_analysis                                                            raise exception
  File "/home/nick/FACT_core/src/storage/db_interface_backend.py", line 188, in _update_analysis
    collection.update_one(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 998, in update_one
    self._update_retryable(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 854, in _update_retryable                                                       return self.__database.client._retryable_write(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/mongo_client.py", line 1491, in _retryable_write
    return self._retry_with_session(retryable, func, s, None)
  File "/usr/local/lib/python3.8/dist-packages/pymongo/mongo_client.py", line 1384, in _retry_with_session
    return func(session, sock_info, retryable)
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 846, in _update
    return self._update(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/collection.py", line 815, in _update
    result = sock_info.command(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/pool.py", line 618, in command
    self._raise_connection_failure(error)
  File "/usr/local/lib/python3.8/dist-packages/pymongo/pool.py", line 603, in command
    return command(self.sock, dbname, spec, slave_ok,
  File "/usr/local/lib/python3.8/dist-packages/pymongo/network.py", line 127, in command
    request_id, msg, size, max_doc_size = message._op_msg(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/message.py", line 706, in _op_msg
    return _op_msg_uncompressed(
OverflowError: MongoDB can only handle up to 8-byte ints
```

Can we write a function that checks if an int is larger than an 8 byte value and if so, make it a string?  If you can provide me with the preferred location for such code I'm happy to submit a pull request.

nstarke, 2020-06-30:
It looks like the `elf_analysis` plugin is causing the issue within the `virtual_address` field as far as I can tell.

jstucke, 2020-07-01:
I have not encountered this problem yet. I will have a look into it.

nstarke, 2020-07-01:
an example value that caused the error is `18446744071578845000`, which is larger than the 64-bit signed integer.  

I tried to write a fix quickly but was not sure of the best place or method to do the checking. My idea was just to cast to string all values for "virtual_address", but again, I'm not sure if that is the best way to go or not.  If you can give me a bit of direction on where and how you want to solve this, I'm happy to submit a PR.

jstucke, 2020-07-01:
In my opinion converting all values to a hex string would be fine (most users would probably prefer a hex representation for addresses anyway). The "virtual_address" values seem to be present both in the "segments" and "sections" data.

PR #449 (2020-07-08): Sd1167 fix 8bit mongo issues
fixes #442
-  values of "virtual_address" and "offset" for sections and segments are converted to hex strings (to fix problems with 8-bit integers and MongoDB)
- fixed collapsing elements in template so that they only collapse when the header is clicked and not the whole table (so that a user can actually select and copy results)
closed 2020-07-08 by jstucke