fkie-cad/FACT_core #444: crypto_material: Template mismatch in ssh.yara

PeterPawn, 2020-07-03:
File:

https://github.com/fkie-cad/FACT_core/blob/master/src/plugins/analysis/crypto_material/signatures/ssh.yara

IMHO the rule `SshRsaPrivateKeyBlock` doesn't consider additional header lines `Proc-Type:` and `DEK-Info:` as used in RSA private key files stored with encryption of the key. The characters `:`, `-`, `,` and `(space)` aren't allowed in the next 32 byte after the header marker.

This assumption is only based on a code review - but it looks like an explanation to me, why your framework didn't recognize the RSA private key from file `etc/default.Fritz_Box_HW236/avm/tr069-default-iad-fritzchen.telekom.de.pem` in the 07.14 firmware for AVM's router model FRITZ!Box 7530 (according to https://github.com/fkie-cad/embedded-evaluation-corpus/blob/master/2020/FKIE-HRS-2020.md it was in the portfolio).

Step(s) to reproduce a file format sample:

```
peh@vidar:~> openssl genrsa -aes128 -passout pass:FooBar 2>/dev/null
-----BEGIN RSA PRIVATE KEY-----
Proc-Type: 4,ENCRYPTED
DEK-Info: AES-128-CBC,2BC62C75AAA7841759672BCAB96420E3

5wCp0dqnwmWD48j+c/VxOLzenJSbP4aQrx7k0MsJeIDqretCAMQ6SXbhuB5eGotN
sQfsbr29FXhPkkrQns2a1OIMIR8aM85f24tvp2GOVEJVsclq+9tYzrW9SlNHE9Wa
0OYq7kQm7FJaqcSLk4HygsO3FEu3W/+ieykZIWDdvAXOGDav01jWA2t8Xrcr9+pQ
3TEwCyANwx+8wsSjvwVeASPXAHj6qbHEPnDl6I0yiLOw0t8x2HV2Ctn3yGa+073d
+cWW1hK1iHtP+bpr3ko8Kbut2kzxx3X/tVNDeneGuHDMDPlvC3pALOwWeCRL3CUt
LIWDCldhrZt9gLloMBYoI/DX5NE5Ippd7nsa6PFBtquv4NoJPXJWd9QuyxuWCJfS
pbpn5SVsVNBnyEeQ3VL6MHdLso8+jcT/YTnDnqSWn2Y6VgwjmpAzkdbFJo/2PQ/F
yzfLHMdTdH2dLcAIIdu7SGlTepzMLVxC2DqyUtliBsteEr3bzMfkjD6VhSvazsov
thWYgQD8/wbppw0DOLtpnOTgi9TeLmiSNgnv3u9sZEY2TtYlZW8jQVVTi6j0L+Jf
kTJB+BUnIEfpcy0IV+uL4R3E/swNouA9vwFff04G6nLF6hgm9vWsGjDKVxlj+Xrf
aK6vrr0y37tPsd2ruyjO03fNt6PwNQEPeXZ+IOs/N86CE9d5NDM44X3WWy4cjdVC
4Juctn8ixBvqxEMgD5kwLAzuXzMQK9KUvTJ7fJVPcCZmoFzR2b8sMKiGKYr+ADCt
fDfbJqNQkLQP2yTBnPzA9QPFbl82aUn7ESJncaPQSsZ6ApCYRdsFinfqF3gYOLIH
dXGw7+GM33A3hv85OUYVMCi7dk+bPeU/Dkphd/G2am7+NoN22b3xedzxlsrUnBL+
VCCP4yZ6/YHnOLD1RoEhz5QV2ydnnTFZeIo/f+RA27QkROE93HXC4lhOgapxDWjC
QNZfs9v6PpcKATAhjHsAUjJkS9eGnyysnA+PHyKXe7AA1aVK3+MdRGuu/jhzzaKK
UTkGIQd57BWO82PIqSa5KHBeBOnldqms6EIg9nJHAc51NsMQowePEt6pt8YSdkdi
k/j3xITTE0CWvzUVsMVNAKt9v1GDphvWqkAzwgIkSE39uMnW5nQPSu1KilYMHO6o
U/maNZa/pthlaIkkLnwWqDF0KdRBH4zJ2wyvm2q/QqT2RmunZf8AHJc7Q1I5n7RH
9cNDOSkHlhnmuRMtEQ70hQAO2aciDthM4WbaO/NYcmWVd3c+fHoD9isFKZwVZRQg
NbM9X6UxHs/HgW5XfbywNzv3AYDKRbC73cwZKPt9/2UAUtGydoQNT+sJMWMVxqvQ
MbzZqauv+qRbK9xVzAl8leExsFM3PP4KsRQYcFM2XedbnPJeHau/d0KPEtwj/zoE
m6DNXBDr0CTL+/7/R+BaIugonVAXF8gmK5hYdGekDrIAMI3PWAd8VJNtzp+K1E0+
0yoyQlXaFRBTFWrjPWhz4KGi4fl8sHwu41auQhpcb7SxcuJMkMjORG6anoZl59Nm
3NzYMHjtvvkMyzX0FKxDehlXXXEli/rcf96VhOrKu0UoLiw/lj/Tdlzmq1pH9p4/
-----END RSA PRIVATE KEY-----
peh@vidar:~>
```

dorpvom, 2020-07-04:
That's a reasonable guess. This can easily be confirmed. I'd say @jstucke didn't have these additional meta information in mind when creating the rule. Would you like to offer an improved rule? Otherwise I'll try to devise one.

PeterPawn, 2020-07-05:
> Would you like to offer an improved rule?

No, at least not yet. Please correct this yourself, if necessary. I'd prefer an additional rule (because such files have a different security level due to the used encryption and should get counted separately), someone else could/would change the existing one.

A file with private key for the same purpose (TR-069 authentication of CPE to (Telekom) ACS for EasySupport) is present in other firmware (FRITZ!OS) versions, too.

I myself am trying hard to reproduce/check your (reported) results for devices by AVM - I'm using other (and some self-made) tools and so far I can't comprehend some of your findings. That's why I'm trying to analyze your framework to find reasons for such differences - without to use it for real.

jstucke, 2020-12-10:
Resolved in #447

closed 2020-12-10 by jstucke