fkie-cad/FACT_core #452: Plugin that outputs dangerous function usages

nstarke, 2020-07-19:
Hi,

Just a suggestion, a plugin which looks at binaries in a standard format (elf, mach-o, pe) and outputs a list of dangerous function usages.  Maybe a count of how many times a dangerous c function was used, or the memory addresses at which the dangerous function is used.  Ghidra is well suited to this task, but other programs may be suitable as well. 

To start with, I would propose using this list for defining "dangerous c functions": https://github.com/intel/safestringlib/wiki/SDL-List-of-Banned-Functions.

Just an idea that I would find helpful personally.

Enkelmann, 2020-07-20:
FACT can already do that. Just run the `cwe checker` analysis on a binary. This reports usages of dangerous functions as instances of CWE 676. It may not work on PE or Mach-O files, though.

Since the `cwe checker` analysis can be quite resource-demanding for larger binaries, it is not included in the default analysis preset. You have to manually select it for it to run.

nstarke, 2020-07-20:
Thank you for the information! Closing this ticket now.

closed 2020-07-20 by nstarke