fkie-cad/FACT_core #459: Incorrect FORTIFY_SOURCE check

disconnect3d, 2020-07-28:
**Tl;dr: When the exploit mitigation finds a binary has stack canaries it will also report that it has fortify source mitigation, which might not be true.**

The exploit mitigation plugin checks the FORTIFY_SOURCE by grepping the output of `readelf` performed on a given binary and looking for the `_chk` string:

https://github.com/fkie-cad/FACT_core/blob/095b7d41f24e5bbb890f7337dae0915305ccd055/src/plugins/analysis/checksec/code/checksec.py#L51-L57

However, when binary has a stack canary, it will (usually) have a symbol named `__stack_chk_fail`, which doesn't relate to the fortify source mitigation:

https://github.com/fkie-cad/FACT_core/blob/095b7d41f24e5bbb890f7337dae0915305ccd055/src/plugins/analysis/checksec/code/checksec.py#L76-L88

This bug can be observed by compiling a `int main() {}` program with `gcc -fstack-protector-all` and `gcc -fno-stack-protector`.

First binary results:
[image]

Second binary results:
[image]

The same bug was also in checksec.sh that I believe the FACT's implementation was based on, which is already fixed: https://github.com/slimm609/checksec.sh/issues/103.

PS: I also described it in https://twitter.com/disconnect3d_pl/status/1288148576317722626.

0xricksanchez, 2020-07-28:
Hi @disconnect3d thanks for reporting in. I can confirm that tricking the FORTIFY check with either a local function with the following naming scheme `XXX_chk` or via the typical stack canary symbol `__stack_chk_fail` is possible. That definitely went under the radar. I'll prepare a fix ASAP.

Also regarding the multitude of other possible exploit mitigations: It definitely would be nice to extend the current plugin implementation to support them too!

disconnect3d, 2020-07-28:
> Also regarding the multitude of other possible exploit mitigations: It definitely would be nice to extend the current plugin implementation to support them too!

You might want to consider delegating the detection plugin to other tools, such as https://github.com/etke/checksec.rs

0xricksanchez, 2020-07-29:
Should be fixed with the merge of #460. Improving the exploit mitigation check will be done in #461.

closed 2020-07-29 by 0xricksanchez