fkie-cad/FACT_core #464: upload failed

weina2016, 2020-08-04:
Hi,
    I found it didn't analysis  when I upload a firmware. I got the uid of the firmware ,but I didn't find any analysis running. It seemed like it didn't upload successfully.It didn't show any error message. So I had to restart the FACT.(version 3.1)
    Look forward for your response.

dorpvom, 2020-08-05:
Hi,
the issue is to abstract to debug without additional information. We are currently observing some issues in combination with firefox, stemming from long response times. Can you reproduce the issue? If so, it'd be great to see the frontend log.

LyleMi, 2020-09-01:
Hi, I also encountered this problem, maybe you could reproduce it by upload a very big firmware (maybe more than 1GB). I tested it on my own virtual machine and can reproduce the problem stably.

jstucke, 2020-09-02:
We will evaluate the use of the suggested JavaScript tool for circumventing the current problems with large firmware uploads. It could take some time until we get to work on this, though.

jstucke, 2021-11-05:
It seems this fixed itself when we switched to Flask 2, therefore I will close this issue. If there are still problems, please reopen it.

closed 2021-11-05 by jstucke