fkie-cad/FACT_core #477: How to repair/uninstall failed installation?

jahatfi, 2020-08-25:
I tried to install FACT on Kali 2020 but it's stuck on the "Install static jstree content" step.  If I can't get it to work I'd like to clean up all the files to recover my storage space, but I don't see an uninstall script.  Where would I find that?

dorpvom, 2020-08-26:
Good question / issue. We haven't thought about an uninstall script yet, and as we have a lot of dependencies, there is no quick way of constructing one.
I'll put that on the to do list, but I might be a while before we get to it.

Meanwhile, can you expand on the jstree problem? 
The internal steps are: (from FACT_core root folder)

```bash
cd src/web_interface/static
wget https://github.com/vakata/jstree/zipball/3.3.9
unzip 3.3.9
rm 3.3.9
rm -rf ./web_js/jstree/vakata*
mv vakata* web_js/jstree
```

jahatfi, 2020-08-26:
It's stuck on the step as mentioned.  I killed it with ctrl+C and here's the output in case it helps:
[image]

`

dorpvom, 2020-08-26:
Quick question: Did you use `root` user for installation and swiched to `user` user afterwards? That would possibly explain the behavior, since the `rm -rf ` would then probably fail. On the other hand this should lead to an error and not a _hanging-idefinitely_.
I'd be interested in which of the commands listed above (wget, unzip, ...) exactly leads to the _hanging_. That's sadly not transparent from the error log due to the implementation of the installation process.

jahatfi, 2020-08-26:
I think I tried both with and without *sudo*.  I think the instructions explicitly say NO sudo, but I tried it after it failed the first time.  I can try to dig into the installation commands later, but at first glance may I recommend including a timeout for your *execute_shell_command* calls?  That should prevent such commands from hanging like this.

dorpvom, 2020-08-26:
A timeout would indeed be help. Especially in this case, where it would offer the information on which command it fails and why. The `execute_shell_command` function does support it. Another item for the to do list (which might be done rather shortly).

jahatfi, 2020-08-26:
I just made those changes and will submit a PR shortly to save you the trouble.

dorpvom, 2020-08-28:
#481 seems to show that indeed the removing of jstree is the issue. One thing you can do, which would be helpful to prevent future issues as well, is `<sudo> chown -R user:user FACT_core/src` (and the same for your data folder, by default that's `/media/data`).
This should fix your permission error. And there really should not be anything not owned by a normal user in the FACT directories.

closed 2022-01-24 by rhelmke