fkie-cad/FACT_core #487: Internal Server Error (http://127.0.0.1:5000)

csenger18, 2020-09-04:
Linux version details:

NAME="Kali GNU/Linux Rolling"
VERSION="2020.3"
KERNEL VERSION="5.5.0-kali1-amd64"
--------------------------------------------------------------
uwsgi version: 2.0.19.1

-------------------------------------------------------------

FACT_core CLI output: 

root@kali:~/FACT_core# ./start_all_installed_fact_components 
[2020-09-04 23:57:06][start_all_installed_fact_components][INFO]: starting db
[2020-09-04 23:57:06][MongoMgr][INFO]: start local mongo database
[2020-09-04 23:57:08][start_all_installed_fact_components][INFO]: starting frontend
[2020-09-04 23:57:08][start_all_installed_fact_components][INFO]: starting backend
[2020-09-04 23:57:09][plugin][WARNING]: Plugin has no code directory: /root/FACT_core/src/plugins/analysis/__pycache__
[uWSGI] getting INI configuration from /root/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.19.1 (64bit) on [Fri Sep  4 23:57:15 2020] ***
compiled with version: 10.2.0 on 03 September 2020 14:21:09
os: Linux-5.5.0-kali1-amd64 #1 SMP Debian 5.5.13-2kali1 (2020-04-03)
nodename: kali
machine: x86_64
clock source: unix
pcre jit disabled
detected number of CPU cores: 4
current working directory: /root/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 47565
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5000 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
Python version: 2.7.18 (default, Apr 20 2020, 20:30:41)  [GCC 9.3.0]
Python main interpreter initialized at 0x561a655cb000
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 26, in <module>
    from common_helper_files import create_dir_for_file
ImportError: No module named common_helper_files
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 252790)
spawned uWSGI worker 1 (pid: 252791, cores: 2)
spawned uWSGI worker 2 (pid: 252792, cores: 2)
spawned uWSGI worker 3 (pid: 252793, cores: 2)
spawned uWSGI worker 4 (pid: 252794, cores: 2)
spawned uWSGI worker 5 (pid: 252795, cores: 2)
*** Stats server enabled on 127.0.0.1:9191 fd: 17 ***
[2020-09-04 23:57:27][cwe_checker][INFO]: Module versions are {'CWE190': '0.1', 'CWE215': '0.1', 'CWE243': '0.1', 'CWE248': '0.1', 'CWE332': '0.1', 'CWE367': '0.1', 'CWE426': '0.1', 'CWE457': '0.1', 'CWE467': '0.1', 'CWE476': '0.3', 'CWE560': '0.1', 'CWE676': '0.1', 'CWE782': '0.1', 'Memory': '0.1'}
[2020-09-04 23:57:28][input_vectors][INFO]: Up and running.
[2020-09-04 23:57:28][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.8/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2020-09-04 23:57:28][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.8/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
[2020-09-04 23:57:28][Analysis][INFO]: Analysis System online...
[2020-09-04 23:57:28][Analysis][INFO]: Plugins available: ['binwalk', 'cpu_architecture', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'dummy_plugin_for_testing_only', 'elf_analysis', 'exploit_mitigations', 'file_hashes', 'file_system_metadata', 'file_type', 'init_systems', 'input_vectors', 'ip_and_uri_finder', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'qemu_exec', 'software_components', 'source_code_analysis', 'string_evaluator', 'tlsh', 'users_and_passwords']
[2020-09-04 23:57:28][analysis_tag][INFO]: Tagging daemon online
[2020-09-04 23:57:28][Unpacking][INFO]: Unpacker Module online
[2020-09-04 23:57:28][Unpacking][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2020-09-04 23:57:29][compare][INFO]: Plug-ins available: ['File_Coverage', 'File_Header', 'Software']
[2020-09-04 23:57:29][Compare][INFO]: Compare Scheduler online...
[2020-09-04 23:57:29][back_end_binding][INFO]: InterCom started

--- no python application found, check your startup logs for errors ---
[pid: 252792|app: -1|req: -1/1] 127.0.0.1 () {32 vars in 502 bytes} [Fri Sep  4 23:57:37 2020] GET / => generated 21 bytes in 0 msecs (HTTP/1.1 500) 2 headers in 83 bytes (0 switches on core 0)

jstucke, 2020-09-08:
> Python version: 2.7.18 (default, Apr 20 2020, 20:30:41) [GCC 9.3.0]

This seems odd. The Application runs with python3. This would also explain why the Import failed

> File "flask_app_wrapper.py", line 26, in
> from common_helper_files import create_dir_for_file
> ImportError: No module named common_helper_files

and why consequently no app was found 

> --- no python application found, check your startup logs for errors ---

I'm not sure why your uWSGI uses Python 2 though. You could try removing your uWSGI and reinstalling through pip:
```bash
python3 -m pip install uwsgi
```

Alternatively, there is a chapter in the uWSGI docs about building it for different Python versions: https://uwsgi-docs.readthedocs.io/en/latest/WSGIquickstart.html#bonus-multiple-python-versions-for-the-same-uwsgi-binary

csenger18, 2020-09-08:
Thanks for the solution, it worked, but I have some struggle with uninstall uwsgi 

after apt-get remove uwsgi and python3 -m unistall uwsgi   I ran python3 -m pip install uwsgi 

command result is: Requirement already satisfied: uwsgi in /usr/local/lib/python3.8/dist-packages (2.0.19.1) but after that I ran 

apt-get remove  uwsgi [TAB] appears uwsgi-core.

After removing uwsgi-core, the python3 -m pip install uwsgi command ran successfully and successfully resolved the issue.

csenger18, 2020-09-08:
because of these I won't   open a new issue just put here
this is what happen at startup:
       ./start_all_installed_fact_components 
        [2020-09-08 22:02:31][start_all_installed_fact_components][INFO]: starting db
        [2020-09-08 22:02:33][start_all_installed_fact_components][INFO]: starting frontend
        [2020-09-08 22:02:33][start_all_installed_fact_components][INFO]: starting backend
        [2020-09-08 22:02:39][MongoMgr][INFO]: start local mongo database
--------> [2020-09-08 22:02:42][plugin][WARNING]: Plugin has no code directory: /root/FACT_core/src/plugins/analysis/__pycache__ 

-----------------------------------------------------------------------------------------------------------------------------------------------

this is what happen at shutdown:  (Crtl+C)
[2020-09-08 22:10:39][start_fact_backend][INFO]: Shutting down components
[2020-09-08 22:10:39][back_end_binding][INFO]: InterCom down
[2020-09-08 22:10:39][Compare][INFO]: Compare Scheduler offline
[2020-09-08 22:10:40][Unpacking][INFO]: Unpacker Module offline
[2020-09-08 22:10:40][analysis_tag][INFO]: Tagging daemon offline
[2020-09-08 22:10:40][Analysis][INFO]: Analysis System offline
------> [2020-09-08 22:10:40][process][ERROR]: SHUTTING DOWN SYSTEM
[2020-09-08 22:10:54][start_fact_db][INFO]: shutting down FACT DB-Service...
[2020-09-08 22:10:56][MongoMgr][INFO]: stop local mongo database
root@kali:~/FACT_core#

jstucke, 2020-09-09:
> [2020-09-08 22:02:42][plugin][WARNING]: Plugin has no code directory: /root/FACT_core/src/plugins/analysi/pycache
> [2020-09-08 22:10:40][process][ERROR]: SHUTTING DOWN SYSTEM

These messages are normal and can safely be ignored

csenger18, 2020-09-09:
Thanks for help. :)

jstucke, 2020-09-09:
I'm glad I could help you with your issue. 😀

closed 2020-09-09 by jstucke