fkie-cad/FACT_core #498: PDF report with details

chfl4gs, 2020-09-30:
The PDF only dumps out firmware collection from mongodb via pdf helper script. File objects collection is completely left out. Is it possible to have file relevant file objects dump out? This is useful if details are included as appendix in PDF document itself.

dorpvom, 2020-10-01:
Hi,

this is possible, in principle, though would need some adjustments in the code. Since it would be overkill to recursively include all file object results, we'd have to add a mechanism to enrich the analyses where it makes sense (such as CVE).
As PDF Reporting is still very much in beta, help to achieve this would be appreciated at this point.

The code where these additions would make sense the most is in the `helperFunctions.pdf._initialize_subfolder` method.
Right now this just `json.dumps` the firmware analysis results. These could be post processed to add the file object results.

dorpvom, 2020-10-01:
Btw. I view different levels of detail in the reporting as enviable, but this goes again back to PDF reporting being in beta, and us not having the necessary time to do sth. about it.

chfl4gs, 2020-10-02:
Thanks for the pointers. Enrichment is definitely needed for file object results to be relevant. I'm trying to understand how this is done in `web_interface.components.analysis_routes`. It is rather complex.