fkie-cad/FACT_core #501: Issues with app

jasj23, 2020-10-02:
Hello, 

I can not start the application, I get the following errors:

!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 256476
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address :5000 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26)  [GCC 9.3.0]
Python main interpreter initialized at 0x5589f7148e50
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 26, in <module>
    from common_helper_files import create_dir_for_file
ModuleNotFoundError: No module named 'common_helper_files'
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***

Please could you give me a  hand? I appreciate it.

jstucke, 2020-10-02:
Hi,

>  ModuleNotFoundError: No module named 'common_helper_files'

Sounds like FACT was not installed correctly. Did you run the `pre_install.sh` and `install.py` scripts?

> *** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***

Also, you should probably not run FACT as root.

jasj23, 2020-10-02:
Hello, 

yes, I ran pre_install.sh and install.py scripts, and everything seems to be ok. 

and if i run FACT as not root I get:

/FACT_core$ ./start_all_installed_fact_components 
[2020-10-02 15:50:26][start_all_installed_fact_components][INFO]: starting db
Traceback (most recent call last):
  File "/home/juan/FACT_core/src/../start_fact_db", line 40, in <module>
    if was_started_by_start_fact():
  File "/home/juan/FACT_core/src/helperFunctions/program_setup.py", line 89, in was_started_by_start_fact
    parent = ' '.join(psutil.Process(os.getppid()).cmdline())
TypeError: 'list' object is not callable
[2020-10-02 15:50:28][start_all_installed_fact_components][INFO]: starting frontend
[2020-10-02 15:50:28][start_all_installed_fact_components][INFO]: starting backend
Traceback (most recent call last):
  File "/home/juan/FACT_core/src/../start_fact_frontend", line 71, in <module>
    if was_started_by_start_fact():
  File "/home/juan/FACT_core/src/helperFunctions/program_setup.py", line 89, in was_started_by_start_fact
    parent = ' '.join(psutil.Process(os.getppid()).cmdline())
TypeError: 'list' object is not callable
Traceback (most recent call last):
  File "/home/juan/FACT_core/src/../start_fact_backend", line 46, in <module>
    if was_started_by_start_fact():
  File "/home/juan/FACT_core/src/helperFunctions/program_setup.py", line 89, in was_started_by_start_fact
    parent = ' '.join(psutil.Process(os.getppid()).cmdline())
TypeError: 'list' object is not callable
[2020-10-02 15:50:30][start_all_installed_fact_components][CRITICAL]: Backend did not start. Shutting down...

thanks for your answer.

jstucke, 2020-10-02:
> parent = ' '.join(psutil.Process(os.getppid()).cmdline())
> TypeError: 'list' object is not callable

It sounds like `psutil.Process().cmdline` returns a list for you. 
Can you print the output of `python3 -m pip show psutil`?
What happens if you run `import psutil, os; psutil.Process(os.getppid()).cmdline()` in a python3 shell?

dorpvom, 2020-10-05:
Does it run as root user or does it fail there too?

jasj23, 2020-10-05:
Hello,

**Can you print the output of python3 -m pip show psutil?**

~/Desktop$ python3 -m pip show psutil
Name: psutil
Version: 1.2.1
Summary: A process and system utilities module for Python
Home-page: http://code.google.com/p/psutil/
Author: Giampaolo Rodola
Author-email: psutil@googlegroups.com
License: License :: OSI Approved :: BSD License
Location: /home/juan/.local/lib/python3.8/site-packages
Requires: 
Required-by: 

**What happens if you run import psutil, os; psutil.Process(os.getppid()).cmdline() in a python3 shell?**

Python 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil, os
>>> psutil.Process(os.getppid()).cmdline() 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'list' object is not callable

Thank you for your help

jstucke, 2020-10-05:
> Name: psutil
> Version: 1.2.1

Wow that version is ancient (from 2013) and it seems like the API changed since then. You could try to install a newer Version with `python3 -m pip install -U psutil` that should fix the error.

jasj23, 2020-10-05:
thnak you , that fixed the problem with psutil, but know i get another problem related with pymongo... this neverends... 

pymongo.errors.ServerSelectionTimeoutError: localhost:27018: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 5f7b24e1849ee645b4550e3e, topology_type: Single, servers: [<ServerDescription ('localhost', 27018) server_type: Unknown, rtt: None, error=AutoReconnect('localhost:27018: [Errno 111] Connection refused')>]>
Traceback (most recent call last):
  File "/home/juan/FACT_core/src/../start_fact_frontend", line 82, in <module>
    work_load_stat = WorkLoadStatistic(config=CONFIG, component='frontend')
  File "/home/juan/FACT_core/src/statistic/work_load.py", line 21, in __init__
    self.db = StatisticDbUpdater(config=self.config)
  File "/home/juan/FACT_core/src/storage/db_interface_statistic.py", line 13, in __init__
    super().__init__(config=config)
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 25, in __init__
    self._authenticate()
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 40, in _authenticate
    self.client.admin.authenticate(user, pw, mechanism='SCRAM-SHA-1')
  File "/usr/local/lib/python3.8/dist-packages/pymongo/database.py", line 1492, in authenticate
    self.client._cache_credentials(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/mongo_client.py", line 775, in _cache_credentials
    server = self._get_topology().select_server(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 241, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 199, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 215, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: localhost:27018: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 5f7b24e1e17e80a8b68a4c1a, topology_type: Single, servers: [<ServerDescription ('localhost', 27018) server_type: Unknown, rtt: None, error=AutoReconnect('localhost:27018: [Errno 111] Connection refused')>]>

I have deleted and reinstalled everything.... but no luck .... 

Thanks a lot for your help again.

jstucke, 2020-10-05:
> pymongo.errors.ServerSelectionTimeoutError: localhost:27018: [Errno 111] Connection refused

That sound like the web frontend has no connection to the database. Did you start FACT with `start_all_installed_fact_components`?
Did you maybe change the password in `main.cfg`? If so, you may need to delete the files in `/media/data/fact_fw_data/` and rerun `python3 src/init_database.py` or else the authentication will fail.

jasj23, 2020-10-05:
Hi, 

yes I start with start_all_installed_fact_components. 

And I did not change anything from main.cfg. 

Maybe it is a pymongo problem or the service is not started well. It is quite extrange... I have had fact installed in other ubuntu versions without problems.... but with 20.04 did not. 

Thanks again for trying to help me

jstucke, 2020-10-06:
Is your MongoDB service running? What does `service mongodb status` return?
If it is not running, you could try starting it with `service mongodb start`.

jasj23, 2020-10-06:
Hello, 

Yes, it is running, once running fact with start_all_installed_fact_components, I get following errors:

Traceback (most recent call last):
  File "/home/juan/FACT_core/src/../start_fact_frontend", line 82, in <module>
    work_load_stat = WorkLoadStatistic(config=CONFIG, component='frontend')
  File "/home/juan/FACT_core/src/statistic/work_load.py", line 21, in __init__
    self.db = StatisticDbUpdater(config=self.config)
  File "/home/juan/FACT_core/src/storage/db_interface_statistic.py", line 13, in __init__
    super().__init__(config=config)
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 25, in __init__
    self._authenticate()
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 40, in _authenticate
    self.client.admin.authenticate(user, pw, mechanism='SCRAM-SHA-1')
  File "/usr/local/lib/python3.8/dist-packages/pymongo/database.py", line 1492, in authenticate
    self.client._cache_credentials(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/mongo_client.py", line 775, in _cache_credentials
    server = self._get_topology().select_server(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 241, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 199, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 215, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: localhost:27018: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 5f7c5c6a6e54c5f021b68aa1, topology_type: Single, servers: [<ServerDescription ('localhost', 27018) server_type: Unknown, rtt: None, error=AutoReconnect('localhost:27018: [Errno 111] Connection refused')>]>
^C[2020-10-06 14:02:52][start_all_installed_fact_components][INFO]: shutting down...

jstucke, 2020-10-06:
> pymongo.errors.ServerSelectionTimeoutError: localhost:27018

Is exactly what happens when MongoDB is not running, though. 
Could you please post the output of `service mongodb status`?
What happens when you start only the database with `python3 src/start_fact_db.py`?
What is the output if you start the database directly with `mongod --config src/config/mongod.conf`?

jasj23, 2020-10-06:
Yes sure:

**service mongodb status**
service mongodb status
● mongodb.service - LSB: An object/document-oriented database
     Loaded: loaded (/etc/init.d/mongodb; generated)
     Active: active (exited) since Mon 2020-10-05 15:41:55 CEST; 24h ago
       Docs: man:systemd-sysv-generator(8)
    Process: 1309 ExecStart=/etc/init.d/mongodb start (code=exited, status=0/SUCCESS)

Oct 05 15:41:53 xxxx-OptiPlex-7050 systemd[1]: Starting LSB: An object/document-oriented database...
Oct 05 15:41:54 xxxx-OptiPlex-7050 mongodb[1309]:  * Starting database mongodb
Oct 05 15:41:55 xxxx-OptiPlex-7050 mongodb[1309]:    ...done.
Oct 05 15:41:55 xxxx-OptiPlex-7050 systemd[1]: Started LSB: An object/document-oriented database.

**What happens when you start only the database with python3 src/start_fact_db.py?**
[2020-10-06 16:32:17][MongoMgr][INFO]: start local mongo database
Traceback (most recent call last):
  File "start_fact_db.py", line 48, in <module>
    work_load_stat = WorkLoadStatistic(config=config, component='database')
  File "/home/juan/FACT_core/src/statistic/work_load.py", line 21, in __init__
    self.db = StatisticDbUpdater(config=self.config)
  File "/home/juan/FACT_core/src/storage/db_interface_statistic.py", line 13, in __init__
    super().__init__(config=config)
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 25, in __init__
    self._authenticate()
  File "/home/juan/FACT_core/src/storage/mongo_interface.py", line 40, in _authenticate
    self.client.admin.authenticate(user, pw, mechanism='SCRAM-SHA-1')
  File "/usr/local/lib/python3.8/dist-packages/pymongo/database.py", line 1492, in authenticate
    self.client._cache_credentials(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/mongo_client.py", line 775, in _cache_credentials
    server = self._get_topology().select_server(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 241, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 199, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.8/dist-packages/pymongo/topology.py", line 215, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: localhost:27018: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 5f7c7ff1c584ee14981ad3cb, topology_type: Single, servers: [<ServerDescription ('localhost', 27018) server_type: Unknown, rtt: None, error=AutoReconnect('localhost:27018: [Errno 111] Connection refused')>]>

**What is the output if you start the database directly with mongod --config src/config/mongod.conf?**
yeap I gets authenticated:
me: "Linux", architecture: "x86_64", version: "5.4.0-48-generic" }, platform: "CPython 3.8.2.final.0" }
2020-10-06T16:35:54.988+0200 I NETWORK  [listener] connection accepted from 127.0.0.1:56922 #116 (66 connections now open)
2020-10-06T16:35:54.988+0200 I NETWORK  [conn116] received client metadata from 127.0.0.1:56922 conn116: { driver: { name: "PyMongo", version: "3.11.0" }, os: { type: "Linux", name: "Linux", architecture: "x86_64", version: "5.4.0-48-generic" }, platform: "CPython 3.8.2.final.0" }
2020-10-06T16:35:54.993+0200 I ACCESS   [conn116] Successfully authenticated as principal fact_admin on admin
****

SO THAT WORKS.... 

**But if I run fact with ./start_all_installed_fact_components, I get again the app error...** 
~/FACT_core$ ./start_all_installed_fact_components 
[2020-10-06 16:35:36][start_all_installed_fact_components][INFO]: starting db
[2020-10-06 16:35:36][MongoMgr][INFO]: start local mongo database
[2020-10-06 16:35:38][start_all_installed_fact_components][INFO]: starting frontend
[2020-10-06 16:35:38][start_all_installed_fact_components][INFO]: starting backend
[2020-10-06 16:35:38][plugin][WARNING]: Plugin has no code directory: /home/juan/FACT_core/src/plugins/analysis/__pycache__
[uWSGI] getting INI configuration from /home/juan/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.19.1 (64bit) on [Tue Oct  6 16:35:39 2020] ***
compiled with version: 9.3.0 on 25 September 2020 11:20:04
os: Linux-5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020
nodename: juan-OptiPlex-7050
machine: x86_64
clock source: unix
detected number of CPU cores: 8
current working directory: /home/juan/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
your processes number limit is 256476
your memory page size is 4096 bytes
[... 7 lines cut ...]
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 26, in <module>
    from common_helper_files import create_dir_for_file
ModuleNotFoundError: No module named 'common_helper_files'
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 50531)
spawned uWSGI worker 1 (pid: 50532, cores: 2)
spawned uWSGI worker 2 (pid: 50533, cores: 2)
spawned uWSGI worker 3 (pid: 50534, cores: 2)
spawned uWSGI worker 4 (pid: 50535, cores: 2)
[... cut ...]
[pid: 50532|app: -1|req: -1/1] 127.0.0.1 () {42 vars in 776 bytes} [Tue Oct  6 16:35:50 2020] GET / => generated 21 bytes in 0 msecs (HTTP/1.1 500) 2 headers in 83 bytes (0 switches on core 0)
--- no python application found, check your startup logs for errors ---
[pid: 50533|app: -1|req: -1/2] 127.0.0.1 () {38 vars in 670 bytes} [Tue Oct  6 16:35:51 2020] GET /favicon.ico => generated 21 bytes in 0 msecs (HTTP/1.1 500) 2 headers in 83 bytes (0 switches on core 0)
[2020-10-06 16:35:53][cwe_checker][INFO]: Module versions are {'CWE190': '0.1', 'CWE215': '0.1', 'C

Thank you again for this great support.

jstucke, 2020-10-06:
> Traceback (most recent call last):
> File "flask_app_wrapper.py", line 26, in
> from common_helper_files import create_dir_for_file
> ModuleNotFoundError: No module named 'common_helper_files'

This should not happen: The package should get installed during the installation. 
What is the output of `python3 -m pip show common_helper_files`?
Do you have multiple versions of python3 installed or are you running FACT in a virtualenv?

jasj23, 2020-10-06:
Sure:

**What is the output of python3 -m pip show common_helper_files?**
~/FACT_core$ python3 -m pip show common_helper_files
Name: common-helper-files
Version: 0.2.2
Summary: file operation helper functions
Home-page: http://www.fkie.fraunhofer.de
Author: Fraunhofer FKIE
Author-email: peter.weidenbach@fkie.fraunhofer.de
License: GPL-3.0
Location: /usr/local/lib/python3.8/dist-packages
Requires: bitmath
Required-by: common-helper-yara, common-helper-process, common-helper-passwords, common-analysis-ip-and-uri-finder

**Do you have multiple versions of python3 installed or are you running FACT in a virtualenv?**
Just Python 3.8.2, and no I am not running FACT in a virtualenv.

jstucke, 2020-10-07:
This is kind of odd: You have the `common_helper_files` module installed but it cannot be found during execution. I'm not entirely sure how this could work. Somehow the python that gets executed by uWSGI does not seem to be the same as the one where the package is installed. I have no idea how to explain this if you don't have other versions installed or a venv, though.

Is the user that you used to install FACT the same as the one that you use to execute FACT? Maybe the package is only installed for one user. At this point I'm not sure how to fix this and would suggest trying to reinstall FACT. Speaking of the installation: Did you restart between running the `pre_install.sh` and `install.py`?

jasj23, 2020-10-08:
yes I did, at this point, I have no idea what is wrong. Ill get back to you if I figure out how to fix it. Thanks again for your great support.

rhelmke, 2021-11-12:
Hey, we are going to close this issue for the time being, as there is few activity in here. In case there are any followup questions regarding this matter, feel free to re-open it.

Cheers.

closed 2021-11-12 by rhelmke