fkie-cad/FACT_core #502: Creation of signatures failed

terorr, 2020-10-12:
./compile_yara_signatures.py 
Create signature directory /home/hjk/FACT_core/src/analysis/signatures
Compile signatures in /home/hjk/FACT_core/src/plugins/analysis/software_components/signatures
Compile signatures in /home/hjk/FACT_core/src/plugins/analysis/crypto_hints/signatures
[ERRROR] Creation of /home/hjk/FACT_core/src/analysis/signatures failed !!
Compile signatures in /home/hjk/FACT_core/src/plugins/analysis/crypto_material/signatures
Compile signatures in /home/hjk/FACT_core/src/plugins/analysis/known_vulnerabilities/signatures

jstucke, 2020-10-14:
Hi, I could not reproduce the error. Do you have a different Yara version installed than 3.7.1? Did you change any of the signatures?
It seems that `src/plugins/analysis/crypto_hints/signatures` caused the error. It would be interesting to see the actual error message. Could you please post the result of 
```yarac -d test_flag=false -w src/plugins/analysis/crypto_hints/signatures/crypto_signatures.yar  /tmp/signature.yc```?

terorr, 2020-10-14:
$ yarac -d test_flag=false -w src/plugins/analysis/crypto_hints/signatures/crypto_signatures.yar /tmp/signature.yc
src/plugins/analysis/crypto_hints/signatures/crypto_signatures.yar(6): error: syntax error, unexpected _LT_
src/plugins/analysis/crypto_hints/signatures/crypto_signatures.yar(21): error: unterminated regular expression

jstucke, 2020-10-14:
> src/plugins/analysis/crypto_hints/signatures/crypto_signatures.yar(21): error: unterminated regular expression

As far as I can tell, there shouldn't even be a regular expression in that line. The file is installed from `https://github.com/Yara-Rules/rules/blob/master/crypto/crypto_signatures.yar`. You could try running `src/plugins/analysis/crypto_hints/install.sh` again. 
Did you perhaps change the file? Can you post the Yara rule from the file which is causing the error message?

terorr, 2020-10-14:
OK，thank you！

dorpvom, 2020-12-09:
Sound like issue is fixed? Re-open otherwise

closed 2020-12-09 by dorpvom