fkie-cad/FACT_core #504: ERROR: child process failed exited with error number 100

acarr91, 2020-10-22:
I have the FACT 3.2-dev version.
Sometimes during the analysis I noticed that it comes in some points and gets stuck until I close the tool to cancel the analysis in progress. From the debug version log I noticed this:

OSError: [Errno 5] Input / output error
-----
ERROR: child process failed, exited with error number 100

and then this message appears in a loop: 
[Unpacking] [DEBUG]  throttle down unpacking to reduce memory consumption ...

This happens only with the analysis of some firmware, others have been completed correctly. What could be the problem/solution?

Thanks

jstucke, 2020-10-23:
The throttling was built in to reduce issues with limited system resources (especially RAM). If the load of the analysis process is too high, the addition of new files to the unpacking / analysis process gets delayed. 
There should be a different reason for the failing child processes. Did you maybe run out of RAM during the analysis? It could be that this only happens during the analysis of larger firmware. If so, you could try to limit the number of worker processes in the `main.cfg`.

acarr91, 2020-10-23:
In the attached image the situation of the system at the time of the block. The 3 components are on a single virtual machine. The analysis has been running like this for more than 12 hours, so I think the machine must be restarted now with the loss of the analysis in progress or is there some way to make the analysis restart from where it stopped?
[image]

jstucke, 2020-10-23:
Sadly, resuming an analysis is currently not possible. Some operations are cached in the database but not the actual files that are currently in the analysis or unpacking process. It's definitely on our agenda for future scheduling improvements. Of course intermediate analysis results up to that point are stored in the database.

acarr91, 2020-11-04:
- I noticed that the errors mentioned in the first post occur if the cwe_checker (mostly) and input vector plugins are present in the analysis. Having 16 cores and 64gb of ram available, could this be due to some unhandled exception in these plugins?

- When an analysis is blocked how can the current analysis be canceled? Is there any way other than rebooting the machine?

- If I had to do a git pull to update the release with the latest updated files on github (hoping that the problem is solved) the application might not work anymore? should I reinstall everything losing the analyzes already carried out?

Thanks for the attention!

jstucke, 2020-11-04:
> If I had to do a git pull to update the release with the latest updated files on github (hoping that the problem is solved) the application might not work anymore? should I reinstall everything losing the analyzes already carried out?

We try to watch out for backwards compatibility. So normally, this should not be an issue and you should not lose any data or analyses. It is important that you keep your passwords from the main config file if you changed them, because they are needed for authorization with the database. If there were some changes to plugins, etc. it should be enough to rerun the `install.py` script with parameters `-B` for backend or `-F` for the frontend. If it runs on a VM you could also make a snapshot to be safe.

> When an analysis is blocked how can the current analysis be canceled? Is there any way other than rebooting the machine?

Sadly, there is currently no other way than shutting down the backend to cancel an analysis. But this is definitely a point on our ToDo list.

> I noticed that the errors mentioned in the first post occur if the cwe_checker (mostly) and input vector plugins are present in the analysis. Having 16 cores and 64gb of ram available, could this be due to some unhandled exception in these plugins?

The cwe_checker plugin does not run 100% stably at the moment. Did you per chance find any stack traces, etc. in the backend output that might let us infer what went wrong?

jstucke, 2020-11-05:
@Enkelmann improved the stability of the cwe_checker plugin in #509 so this should hopefully be less of an issue.

acarr91, 2020-11-09:
After performing the `git pull`, starting from 3.2-dev (about August), and without launching the `install.py`, I tried to perform an update analysis on a firmware by adding the cwe_checker plugin (where errors had already occurred and the analysis block). Also in this case the same event occurred, but I noticed some unusual errors regarding the cwe checker that I am attaching in the image (piece of the log with grep). There are also some permission denied errors that I do not understand because I have verified the credentials in main.cfg. Anyway I proceed to do an `install.py -B` to check if there are any changes about what is written.
[image]

Enkelmann, 2020-11-09:
I can chime in on the cwe_checker errors:
- The errors with `objdump` are cases where the cwe_checker internally runs objdump on the file to be analyzed but fails to parse the output (usually because objdump returns an actual error). These are cases that currently cannot be analyzed with the cwe_checker.
- In general timeout errors are expected to happen for large files when using the cwe_checker plugin. However, according to the log message it probably was the wrong timeout that stopped the cwe_checker. We will have a look at that.

jstucke, 2020-11-09:
the `PermissionError` could be due to a change in the unpacking process (we moved the owner change of the unpacked files inside the docker container to remove the need for elevated privileges). This should be fixed when you pull the latest version of the FACT extractor:
```docker pull fkiecad/fact_extractor:latest```

acarr91, 2020-11-09:
@Enkelmann should I do a `docker pull fkiecad/cwe_checker:latest` regarding the cwe_checker plugin, as suggested by @jstucke about the extractor?

Enkelmann, 2020-11-09:
That could reduce the error messages from the cwe_checker. However, if the timeout errors from the cwe_checker are the culprit for your problems, then this will probably not help. Ensuring that the cwe_checker gets shut down correctly is the responsibility of the FACT plugin, not of the cwe_checker docker image.

Enkelmann, 2020-11-10:
I have adjusted the internal cwe_checker timeout in #514 (will be merged later today) to allow for a graceful shutdown of the cwe_checker plugin if a timeout occurs. However, in my tests the shutdown seemed to happen gracefully even before the change (just with the wrong error message).

The cwe_checker could still be the culprit if too many instances of it run at the same time. The reason then would be that your system runs out of RAM. If you want to test that hypothesis, you could try to only run the cwe_checker analysis plugin on your firmware sample and see what happens. If this is the culprit, the only workaround right now is to limit the number of worker processes as @jstucke suggested.

rhelmke, 2021-11-12:
Hey,

since there is no more activity in here, I'm going to close this issue. Feel free to re-open it in case the problem still persists.

closed 2021-11-12 by rhelmke