fkie-cad/FACT_core #512: Large disk space occupied after firmware removal

acarr91, 2020-11-06:
After removing some firmware, the space occupied on the disk is still huge.

I have verified that the fact_wt_mongodb folder is 65gb large despite deleting all firmware (except one of 80mb) through the web interface.

(the virtual machine is dedicated to FACT)

Thanks for the attention

[image]
[image]

dorpvom, 2020-11-09:
Hi, I think I've encountered similar behavior in the past as well. I simply put it as a MongoDB quirk and haven't investigated further. I'm pretty sure I found some MongoDB functionality to free unused space though. I'll look for it and get back to you once I found it.

dorpvom, 2020-11-09:
This Thread describes some Mongo stuff:

https://stackoverflow.com/questions/11617862/how-to-reclaiming-deleted-space-without-db-repairdatabase

jstucke, 2020-11-09:
What is also possible is that not all files are deleted yet: because of the way the delete operations are routed (over the "intercom" between the frontend and the backend), the operations are relatively slow (only a couple per second). Therefore, it can take some time to fully delete all files from a firmware. The operations are buffered using the database, though. So when you restart FACT they should continue (they should show up in the logging output).

acarr91, 2020-11-09:
@dorpvom Thanks for the reply. I would like to use the "compact" command, but I am not being able to access mongo with the credentials present in the fact main.cfg

@jstucke Thanks for the reply. I don't think that's the reason because I was monitoring the log (debug mode) in real time and I checked the disk space only after the delete messages have finished.

dorpvom, 2020-11-09:
@acarr91 You can start the database manually and without authentication with `mongod --config src/config/mongod.conf`. Then you can interact with it using `mongo --port 27018` without credentials.

dorpvom, 2020-11-09:
By typing `show dbs` you then also get the sizes of your database, for furher debugging, if necessary.

acarr91, 2020-11-09:
@dorpvom `show dbs` shows that the faf_sanitize database is 62.905GB in size. The other db larger than 0.000GB are fact_main with 0.632GB, fact_intercom_file_delete_task with 0.326GB and fact_intercom_tar_repack_task_resp with 0.142GB. These appear to be normal size. Instead, what does the faf_sanitize db refer to? Could `compact` be the solution to this problem?

dorpvom, 2020-11-09:
It might, though I can't be sure.

The sanitize database is used to store large analysis results (> 100 kB). This is necessary since MongoDB has an upper limit on size for each entry (~ 18 MB). I just checked that these entries should be deleted correctly once the matching firmware is deleted, but I'll do some testing to see if it actually works.

acarr91, 2020-11-09:
Then it could have happened that the removal of the entries in faf_sanitize did not take place at the same time as the removal of the firmware using the web gui. If this behavior is confirmed, would this be a bug to fix?

dorpvom, 2020-11-09:
Yes, that would definitely be something to fix

dorpvom, 2020-11-10:
I'm getting more and more confused by this. So I have checked again, that the implementation of the deletion is sound. Still my local db shows a huge number of files in the sanitize db that belongs to deleted firmware. I've written a short script to check which files belong and which do not:

```python3
from pymongo import MongoClient
from gridfs import GridFS

client = MongoClient('mongodb://127.0.0.1:27018', connect=False)
fw_collection = client['fact_main']['firmwares']
fo_collection = client['fact_main']['file_objects']
sandb = GridFS(client['faf_sanitize'])
data = sandb.find()
yes, no = 0, 0

for entry in sandb.find():
    sha, size = entry.filename.split('_')[-2:]
    uid = f'{sha}_{size}'
    query = {'_id': uid}
    if not fw_collection.find_one(query, {'_id': 1}) and not fo_collection.find_one(query, {'_id': 1}):
        # print(f'[INFO]      {uid} would not be deleted')
        no += 1
    else:
        # print(f'[WARNING]   {uid} would be deleted')
        yes += 1
        # sandb.delete(entry._id)

print(f'\nIn total that would mean {yes} deletions and {no} remaining files in grid.')
```

This assumes your database is up and running on localhost without authentication (see command in previous comment).
Now you can remove the comment `#` in front of the deletion and actually go ahead and delete the entries which are not associated with an existing analysis anymore (you can also remove the comment marks before the logging first to see what is actually removed).

Now I did this and the db size was unchanged afterwards. What's more I additionally run **mongod** with the `--repair` option which should free disk space and it is still unchanged. That does not really play with the fact that my clear up script did 64280 deletions with 813 remaining entries (by the way, running it again it finishes quickly telling me there is nothing to delete, so the entries seem to actually be gone).

I'm at a loss for now, but I'll keep looking for a resolution once I have some time to dig deeper.

dorpvom, 2020-11-10:
Alas. I swapped both cases and now my database has to be restored. Anyway I fixed above code, it's now correct again.

dorpvom, 2020-11-10:
My next move was to basically clone the database. So I wrote a new script to move all data from the old 'faf_sanitize' to a new 'fact_sanitize' database. The new one is factor 100 smaller than the old one...

```python
from pymongo import MongoClient
from gridfs import GridFS

client = MongoClient('mongodb://127.0.0.1:27018', connect=False)
old_san_db = GridFS(client['faf_sanitize'])
new_san_db = GridFS(client['fact_sanitize'])

for entry in old_san_db.find():
    sha, size = entry.filename.split('_')[-2:]
    uid = f'{sha}_{size}'
    query = {'_id': uid}
    # old_san_db.delete(entry._id)
    new_san_db.put(entry.read(), filename=entry.filename)
```

This recovers your disk space if you then set the relevant config entry to `fact_sanitize` and delete the old database. It still does not solve the original problem though.

jstucke, 2020-11-11:
In the course of this issue, we noticed that there was an error in the `fact_sanitize` database where updated analysis entries did not replace old ones, but instead created new duplicate entries. #515 should hopefully fix this issue. It does not delete all duplicate entries, though. Therefore cleaning the database would still be a good idea.

This script should remove the duplicates:
```python
from pymongo import MongoClient
from gridfs import GridFS

client = MongoClient("mongodb://127.0.0.1:27018", connect=False)
sandb = GridFS(client["faf_sanitize"])

filename_list = sorted(filename for filename in sandb.list())
print(f"found {len(filename_list)} different file entries")
for filename in sorted(filename for filename in sandb.list()):
    newest = sandb.get_last_version(filename)
    duplicate_count = sandb.find({"filename": filename, "md5": {"$ne": newest.md5}}).count()
    print(f"file {filename}: {duplicate_count} entries")
    for duplicate in sandb.find({"filename": filename, "md5": {"$ne": newest.md5}}):
        sandb.delete(duplicate._id)
client["faf_sanitize"].command({"compact": "fs.files"})
client["faf_sanitize"].command({"compact": "fs.chunks"})
```

acarr91, 2020-11-16:
@jstucke I did a git pull after merge #515 and ran the last script you posted twice to remove duplicates from `faf_sanitize`. I noticed that the number of entries found in both cases in the first string of the script is the same, while for each entry during the first launch there were nonzero numbers (duplicates), at the second launch all 0. Despite this, I didn't notice any differences in the size of the `faf_sanitize` db and in the space available in the gui's tab system. I didn't understand if I have to perform all the db duplication steps performed by @dorpvom to solve the problem. In that case, I would not want to configure `fact_sanitize` db as official (now `faf_sanitize`) in order to not have compatibility problems with possible future updates.

jstucke, 2020-11-17:
The delete operations are cached and are only really executed when you call [compact](https://docs.mongodb.com/manual/reference/command/compact/). I added this command to the end of the last script and my local `/media/data/fact_wt_mongodb` folder went down in size from 6.5GB to 1.5GB (I had many duplicate entries from repeating analyses during development). If this does not work, you could try calling "compact" in the the mongo shell.

A way to use the migration script from @dorpvom but not losing the original DB name would be to delete the old DB with [dropDatabase()](https://docs.mongodb.com/manual/reference/method/db.dropDatabase/index.html) and copy the new one with [copyDatabase()](https://docs.mongodb.com/manual/reference/method/db.copyDatabase/index.html) to the old name (not very elegant but it seems like there is no other way to rename the DB).

dorpvom, 2020-12-09:
I'm going to close this in the next days as I think we fixed the multiple issues leading to the problem. @acarr91 is there sth. left to do from your point of view?

acarr91, 2020-12-09:
@dorpvom Thanks for the question. I could not fully test the solutions because I was urgently engaged on issues outside FACT, in the next few days I should have the opportunity to test everything. Anyway, theoretically, reading the scripts and solutions provided I think there should be no problems.

dorpvom, 2021-05-06:
Hi @acarr91. I'll close this issue due to inactivity. If the issue persists, feel free to re-open.

closed 2021-05-06 by dorpvom