fkie-cad/FACT_core #525: Error setting up cve_lookup plugin

SonicFM, 2020-12-10:
Hi :)
first of all i want to thank you for this great project! I really appreciate your work!

I hope someone can help me with this.
I am installing FACT via "git clone -b 3.1 --single-branch ...." (so stable release 3.1 not master/develop branch) but when running the install script with "-NU" options (even without) i get the following error and dont know how to solve it.

for clarification in advance:
- preinstall script was run previously and no error occured
- rebooted after this
- there is enough diskspace
- there is enough memory
- internet connection is stable

Error:
------------------------------------
   install cve lookup dependencies
------------------------------------
Requirement already satisfied: pyxdameraulevenshtein in /usr/local/lib/python3.7/dist-packages (1.6.1)
Requirement already satisfied: numpy>=1.16.1 in /usr/local/lib/python3.7/dist-packages (from pyxdameraulevenshtein) (1.19.4)
Setting up database
Traceback (most recent call last):
  File "setup_repository.py", line 297, in <module>
    main()
  File "setup_repository.py", line 288, in main
    init_repository(extraction_path, args.target, years=years)
  File "setup_repository.py", line 231, in init_repository
    import_cve(cve_extract_path=extraction_path, years=years)
  File "setup_repository.py", line 164, in import_cve
    cve_list, summary_list = get_cve_import_content(cve_extract_path, year_selection)
  File "setup_repository.py", line 95, in get_cve_import_content
    dp.download_cve(cve_extraction_path, years=year_selection)
  File "/opt/FACT_core/src/plugins/analysis/cve_lookup/internal/data_parsing.py", line 47, in download_cve
    process_url(url, download_path)
  File "/opt/FACT_core/src/plugins/analysis/cve_lookup/internal/data_parsing.py", line 35, in process_url
    zipped_data = ZipFile(BytesIO(request.content))
  File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

I think it fails downloading and therefore leaving a corrupted or non existend file witch causes the "not a zip file" error.

Thanks for any help or suggestion in advance!

jstucke, 2020-12-11:
If I remember correct, I saw this also happen on our CI but only once in a while. Could you try rerunning the installation and see if it happens again? You can also run the installation of the cve_lookup plugin separately by running `src/plugins/analysis/cve_lookup/install.sh`.

SonicFM, 2020-12-11:
Hi @jstucke,
thank you for you quick reply!

I already did this multiple times, sry i didn't mentioned it.
However the error remains the same.

I also reinstalled the OS multiple times but the error keeps reoccurring.
Also tried different OSs (Debian 10 and Ubuntu/Ubuntu-Server 20.04).

The only way i could set it up, was by replacing the new "built-in" cve-lookup plugin with the old plugin. By downloading the old plugin and replacing the "new" one with the old one in src/plugins/analysis/.

But this was only for testing. As you can image i would love to use the "built-in" plugin ;)

jstucke, 2020-12-11:
Do you have anything in place that could interfere with the download (like proxies, firewalls, etc.)? Since our Ubuntu 20.04 CI build ran successfully less than one hour ago, this at least does not seem to be a problem with the download URLs.

SonicFM, 2020-12-11:
Nope i don't have an firewall which could interfere. My Proxmox does have one but i already disabled the firewall prior due to debugging. The error still persists :(

jstucke, 2020-12-11:
I'm not really sure what the problem could be. Could you try running this script in the FACT src directory and post the result?

```python
import logging
import requests

from plugins.analysis.cve_lookup.internal.data_parsing import process_url, get_cve_links, CVE_URL

for url in get_cve_links(CVE_URL):
    response = requests.get(url, allow_redirects=True)
    try:
        response.raise_for_status()
    except Exception as err:
        logging.error(err, exc_info=True)
    print(f"url: {url}, status_code: {response.status_code}, content: {len(response.content)}")

```

SonicFM, 2020-12-11:
Getting the following:

python3 /opt/FACT_core/src/test.py

ERROR:root:404 Client Error: Not Found for url: https://nvd.nist.gov/feeds/json/cve/1.0/nvdcve-1.0-2020.json.zip
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    response.raise_for_status()
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://nvd.nist.gov/feeds/json/cve/1.0/nvdcve-1.0-2020.json.zip
url: https://nvd.nist.gov/feeds/json/cve/1.0/nvdcve-1.0-2020.json.zip, status_code: 404, content: 0

Also asked a friend of mine. He is also getting "404" or blank-page when he tries to download via Browser.

jstucke, 2020-12-11:
Indeed the links are wrong: they changed to version 1.1 in september (e.g. https://nvd.nist.gov/feeds/json/cve/1.1/nvdcve-1.1-2020.json.zip)
We fixed this problem but sadly, we did not port back the fix to the last stable version which explains your problems. We will probably release a new stable version in the near future to fix this but up until then I would suggest that you either use the latest master version (which should also be stable) or simply fix the link in the file `src/plugins/analysis/cve_lookup/internal/data_parsing.py` from 
```
 CVE_URL = 'https://nvd.nist.gov/feeds/json/cve/1.0/nvdcve-1.0-{}.json.zip' 
```
to
```
 CVE_URL = 'https://nvd.nist.gov/feeds/json/cve/1.1/nvdcve-1.1-{}.json.zip' 
```

SonicFM, 2020-12-11:
Great news, thank you!

I'll report back, if there are any other issues regarding this topic. :)

closed 2020-12-11 by SonicFM