fkie-cad/FACT_core #534: Installation Fail on Installing python2 pip

ruotongyu, 2021-01-28:
The pre-install steps are finished. The installation failed when I run install.py. There is error in pip installation for python2. It seems pip does not python2 sometime recently.

JK167, 2021-01-29:
Got the same problem 
[2021-01-29 00:51:35][common][INFO]: Installing python2 pip
Traceback (most recent call last):
  File "/home/kali/FACT_core/src/./install.py", line 183, in <module>
    install()
  File "/home/kali/FACT_core/src/./install.py", line 164, in install
    common(distribution)
  File "/home/kali/FACT_core/src/install/common.py", line 78, in main
    install_pip('python2')
  File "/home/kali/FACT_core/src/install/common.py", line 20, in install_pip
    raise InstallationError('Error in pip installation for {}:\n{}'.format(python_command, output))
helperFunctions.install.InstallationError: Error in pip installation for python2:
Traceback (most recent call last):
  File "get-pip.py", line 24226, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpI9CWBD/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

0xricksanchez, 2021-01-29:
@ruotongyu @JK167 FACT requires at least python 3.6+ to operate. The stack trace shows that it crashes at the f-string formatting that was introduced in python 3.6 (`sys.stderr.write(f"ERROR: {exc}")`)

In your case @JK167: Even Kali Linux should have fully switched to a recent python3 version by now. 

There won't be any python2 support as large portions of the code specifically utilize the new python 3.6+ functionalities. Furthermore, python2 is EOL for a while already.

PM-1-1, 2021-01-29:
It seems that there is a problem with line 73 in src/install/common.py

 `   install_pip('python2')`

when we look at line 17 install_pip function whe have line:
```

for command in ['wget https://bootstrap.pypa.io/get-pip.py', 'sudo -EH {} get-pip.py'.format(python_command), 'rm get-pip.py']:
        output, return_code = execute_shell_command_get_return_code(command)

```
so get-pip.py is executed by python2

JK167, 2021-01-29:
Yes I added comments # in this line and all went though. Successfully installed

dorpvom, 2021-01-29:
Thanks for the detailed reports. We'll check if we can drop ol' python2 altogether. @jstucke is looking into this right now.

jstucke, 2021-01-29:
It seems like the installation of python2 was just some old artifact and can be safely removed. At least the CI builds passed for #535

PR #535 (2021-01-29): removed python2 installation
resolves #534 
- removed deprecated python2 installation from install script
closed 2021-01-29 by dorpvom