fkie-cad/FACT_core #538: Install on Ubuntu 20.04.2 Not functional - Internal Server Error

sperglord8008s, 2021-02-08:
GUI works and can move around the interface but Upload and tools dont work. 
Currently running Ubuntu in VM from Windows machine. 

~/FACT_core/start_all_installed_fact_components
[2021-02-07 16:45:09][start_all_installed_fact_components][INFO]: starting db
[2021-02-07 16:45:09][MongoMgr][INFO]: start local mongo database
[2021-02-07 16:45:11][start_all_installed_fact_components][INFO]: starting frontend
[2021-02-07 16:45:11][start_all_installed_fact_components][INFO]: starting backend

**[2021-02-07 16:45:11][plugin][WARNING]: Plugin has no code directory: /home/user/FACT_core/src/plugins/analysis/__pycache__
Traceback (most recent call last):**
  File "/home/user/FACT_core/src/../start_fact_backend", line 54, in <module>
    analysis_service = AnalysisScheduler(config=config)
  File "/home/user/FACT_core/src/scheduler/Analysis.py", line 37, in __init__
    self.load_plugins()
  File "/home/user/FACT_core/src/scheduler/Analysis.py", line 212, in load_plugins
    plugin = source.load_plugin(plugin_name)
  File "/usr/local/lib/python3.8/dist-packages/pluginbase.py", line 301, in load_plugin
    return __import__(self.base.package + '.' + name,
  File "/usr/local/lib/python3.8/dist-packages/pluginbase.py", line 438, in plugin_import
    return self._system_import(import_name, globals, locals,
  File "/home/user/FACT_core/src/plugins/analysis/cve_lookup/code/cve_lookup.py", line 12, in <module>
    from pyxdameraulevenshtein import damerau_levenshtein_distance as distance  # pylint: disable=no-name-in-module
  File "/usr/local/lib/python3.8/dist-packages/pluginbase.py", line 438, in plugin_import
    return self._system_import(import_name, globals, locals,
  File "pyxdameraulevenshtein.pyx", line 1, in init pyxdameraulevenshtein

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
[uWSGI] getting INI configuration from /home/user/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.19.1 (64bit) on [Sun Feb  7 16:45:12 2021] ***
compiled with version: 9.3.0 on 07 February 2021 23:00:04
os: Linux-5.8.0-41-generic #46~20.04.1-Ubuntu SMP Mon Jan 18 17:52:23 UTC 2021
nodename: ubuntu
machine: x86_64
clock source: unix
detected number of CPU cores: 8
current working directory: /home/user/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
your processes number limit is 63711
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5000 fd 3
Python version: 3.8.5 (default, Jul 28 2020, 12:59:40)  [GCC 9.3.0]
Python main interpreter initialized at 0x5648ca590f50
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500064 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
[2021-02-07 16:45:13][frontend_main][INFO]: Web front end online
WSGI app 0 (mountpoint='') ready in 1 seconds on interpreter 0x5648ca590f50 pid: 3532 (default app)
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 3532)
spawned uWSGI worker 1 (pid: 3537, cores: 2)
spawned uWSGI worker 2 (pid: 3539, cores: 2)
spawned uWSGI worker 3 (pid: 3541, cores: 2)
spawned uWSGI worker 4 (pid: 3543, cores: 2)
spawned uWSGI worker 5 (pid: 3544, cores: 2)
*** Stats server enabled on 127.0.0.1:9191 fd: 18 ***
[2021-02-07 16:46:38][update][INFO]: time to create stats: 0s
[pid: 3537|app: 0|req: 1/1] 127.0.0.1 () {32 vars in 523 bytes} [Sun Feb  7 16:46:38 2021] GET / => generated 14488 bytes in 56 msecs (HTTP/1.1 200) 2 headers in 82 bytes (1 switches on core 0)
[pid: 3539|app: 0|req: 2/2] 127.0.0.1 () {34 vars in 573 bytes} [Sun Feb  7 16:46:38 2021] GET /ajax/stats/general => generated 41 bytes in 19 msecs (HTTP/1.1 200) 2 headers in 71 bytes (1 switches on core 0)
[2021-02-07 16:46:38,648] ERROR in app: Exception on /ajax/stats/system [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/dist-packages/flask_restful/__init__.py", line 272, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.8/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "./web_interface/components/ajax_routes.py", line 135, in _get_system_stats
    'backend_cpu_percentage': '{}%'.format(backend_data['system']['cpu_percentage']),

TypeError: 'NoneType' object is not subscriptable
[2021-02-07 16:46:38][app][ERROR]: Exception on /ajax/stats/system [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/dist-packages/flask_restful/__init__.py", line 272, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.8/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "./web_interface/components/ajax_routes.py", line 135, in _get_system_stats
    'backend_cpu_percentage': '{}%'.format(backend_data['system']['cpu_percentage']),
TypeError: 'NoneType' object is not subscriptable
[pid: 3539|app: 0|req: 2/3] 127.0.0.1 () {34 vars in 571 bytes} [Sun Feb  7 16:46:38 2021] GET /ajax/stats/system => generated 290 bytes in 22 msecs (HTTP/1.1 500) 2 headers in 84 bytes (1 switches on core 1)
[2021-02-07 16:46:40,554] ERROR in app: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/dist-packages/flask_restful/__init__.py", line 272, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.8/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "./web_interface/components/io_routes.py", line 50, in _app_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "./intercom/front_end_binding.py", line 36, in get_available_analysis_plugins
    raise Exception("No available plug-ins found. FACT backend might be down!")

Exception: No available plug-ins found. FACT backend might be down!

Any help or is Hardware Polling kaput in VM?

jstucke, 2021-02-08:
There seems to be an incompatibility between the versions of the packages pyxdameraulevenshtein and numpy. Could you please print the output of the command `python3 -m pip freeze | grep -iE "levenshtein|numpy"`?

Regardless of that, there seems to be a bug in the live stats functionality on the "home" page.

jstucke, 2021-02-08:
> is Hardware Polling kaput in VM?

FACT should run in a VM without problems

jstucke, 2021-02-08:
After looking a bit into the problem I am fairly positive that your `numpy` version is too old for the installed version of `pyxdameraulevenshtein`. You could either try to downgrade the version of `pyxdameraulevenshtein` or upgrade your `numpy` version.

minus-c, 2021-05-02:
Hi, I am having this same issue on an Ubuntu 18.04 vm, is there a specific numpy that I should be downgrading to?  My output from the command you gave is numpy==1.195 pyxDamerauLevenshtein==1.7.0.  Both appear to be the most up to date in pip3

jstucke, 2021-05-03:
Hi, I'm not sure if there is a version combination that definitely works, but I got it to work in a 18.04 docker container with numpy-1.19.5 and pyxDamerauLevenshtein-1.6.1. Please make sure that you have `gcc` installed, because the pyxDamerauLevenshtein package needs to be compiled.

rhelmke, 2021-11-12:
Hey, due to the inactivity in this issue we are going to close it for the time being. In case there are any followup questions regarding this matter, feel free to re-open it.

Cheers.

closed 2021-11-12 by rhelmke