fkie-cad/FACT_core #540: Scale Analysis

opgeven, 2021-02-16:
Feature request: ability to farm out the analysis to multiple machines to speed up analysis.

jstucke, 2021-02-17:
We have this feature on our agenda, but this is not something we will be able to implement in the short term, sadly.

Theoretically, it could be possible to simply start the backend on multiple machines with the same database and frontend (because the backend uses the database as queue for new tasks and the analysis results are stored in the database). But since the actual files are stored on the backend machine, there would definitely be some bugs (e.g. retrieving the binaries, updating the analysis, deleting firmware, etc.). But since we never tried this, there is no guarantee this actually works. Also there is obviously no load balancing, so which instance of the backend instances grabs the task would be random. You could try it and report your findings.

dorpvom, 2021-02-25:
Small addition to @jstucke: The backend / file storage problem could in theory be solved by using a shared storage filesystem like a ZFS. This would than introduce I/O latency that possibly mitigates the speed up in analysis introduced by the scaling ...