fkie-cad/FACT_core #543: Failed to build on latest Kubuntu. strncpy, multiple definition and ld errors.

mzombor, 2021-02-23:
Hi there! 

I was trying to build and use FACT on my computer which has the latest Kubuntu 20.10 on it. I can provide more version details if needed.
Basically, the build fails with the provided output. Sorry if I'm not really helpful with providing information but I'm not sure what details might be needed. Everything was going well until this happened:

```
...
Configured for building John the Ripper jumbo:

Target CPU ................................. x86_64 AVX2, 64-bit LE
AES-NI support ............................. run-time detection
Target OS .................................. linux-gnu
Cross compiling ............................ no
Legacy arch header ......................... x86-64.h

Optional libraries/features found:
Memory map (share/page large files) ........ yes
Fork support ............................... yes
OpenMP support ............................. no
OpenCL support ............................. yes
Generic crypt(3) format .................... yes
libgmp (PRINCE mode and faster SRP formats)  yes
128-bit integer (faster PRINCE mode) ....... yes
libz (pkzip and some other formats) ........ yes
libbz2 (gpg2john extra decompression logic)  yes
libpcap (vncpcap2john and SIPdump) ......... yes
OpenMPI support (default disabled) ......... no
ZTEX USB-FPGA module 1.15y support ......... no

Install missing libraries to get any needed features that were omitted.

Configure finished.  Now "make -s clean && make -sj4" to compile.
ar: creating aes.a
ar: creating ed25519-donna.a
ar: creating secp256k1.a
In file included from /usr/include/string.h:519,
                 from SIPdump.c:25:
In function ‘strncpy’,
    inlined from ‘parse_payload’ at SIPdump.c:415:5:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: warning: ‘__builtin_strncpy’ specified bound depends on the length of the source argument [-Wstringop-overflow=]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
SIPdump.c: In function ‘parse_payload’:
SIPdump.c:412:9: note: length computed here
  412 |         strlen(conn_table[i].buffer) + 1;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

After this, I get about a bazillion `multiple definition of ...` errors and the output ends with.

```
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1724: ../run/john] Error 1
make: *** [Makefile:190: default] Error 2
```

I have absolutely ZERO idea what could be the problem and I believe posting here would be more useful than spending however many hours trying to solve the problem myself. (I'm on a timer here).

If this is a trivial problem, sorry for wasting your time. I'd be grateful for every bit of help you can give me :)

jstucke, 2021-02-24:
It is the installation of John the Ripper jumbo that seems to be the problem. Could you post the exact error message that led to the linking error ("collect2: error: ld returned 1 exit status")? 
Also, you are probably not the only one having this problem so you could look through their issues if there is anything related 
https://github.com/openwall/john/issues?q=is%3Aissue

If you don't need the password cracking abilities, you can also comment out the John installation in `src/plugins/analysis/users_and_passwords/install.sh`. The rest will hopefully install without problems.

mzombor, 2021-02-25:
Hi there!

First, thank you for your help! :)

Unfortunately, I couldn't find any problems. The fact is, I cloned and built the latest, bleeding edge John jumbo and everything worked just fine. Somehow the install script breaks. The whole error message (with sensitive data HOPEFULLY) redacted is this:

```
2021-02-25 10:56:09 (6,10 MB/s) - ‘1.9.0-Jumbo-1.tar.gz’ saved [47523457]

checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking whether to compile using MPI... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking additional paths...  -L/usr/local/lib -I/usr/local/include
checking arg check macro for -m with gcc... yes
checking arg check macro for -Q with gcc... yes
checking if gcc supports -funroll-loops... yes
checking if gcc supports -Os... yes
checking if gcc supports -finline-functions... yes
checking if gcc supports -Og... yes
checking if gcc supports -Wall... yes
checking if gcc supports -Wno-stringop-truncation... yes
checking if gcc supports -Wno-format-overflow... yes
checking if gcc supports -Wno-format-truncation... yes
checking if gcc supports -Wno-tautological-constant-out-of-range-compare... no
checking if gcc supports -fno-omit-frame-pointer... yes
checking if gcc supports --param allow-store-data-races=0... no
checking if gcc supports -Wno-deprecated-declarations... yes
checking if gcc supports -Wformat-extra-args... yes
checking if gcc supports -Wunused-but-set-variable... yes
checking if gcc supports -Qunused-arguments... no
checking if gcc supports -std=gnu89... yes
checking if gcc supports -Wdate-time... yes
checking if gcc supports -m64 w/ linking... yes
checking for 32/64 bit... 64-bit
checking additional paths (64 bit)...  -L/usr/lib64 -L/lib64
checking whether ln -s works... yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for a sed that does not truncate output... /usr/bin/sed
checking for GNU make... make
checking whether make sets $(MAKE)... yes
checking how to run the C preprocessor... gcc -E
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for sort... /usr/bin/sort
checking for find... /usr/bin/find
checking for perl... /usr/bin/perl
checking for ar... ar
checking for strip... strip
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking if pkg-config will be used... no
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
[... 315 lines cut ...]
/usr/bin/ld: opencl_pgpsda_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_pgpsda_fmt_plug.c:67: multiple definition of `outsize'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
/usr/bin/ld: opencl_pgpwde_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_pgpwde_fmt_plug.c:71: multiple definition of `settingsize'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
/usr/bin/ld: opencl_pgpwde_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_pgpwde_fmt_plug.c:71: multiple definition of `insize'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
/usr/bin/ld: opencl_pgpwde_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_pgpwde_fmt_plug.c:71: multiple definition of `outsize'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
/usr/bin/ld: opencl_pgpwde_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_pgpwde_fmt_plug.c:71: multiple definition of `cracked_size'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
/usr/bin/ld: opencl_sappse_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_sappse_fmt_plug.c:73: multiple definition of `insize'; opencl_agilekeychain_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_agilekeychain_fmt_plug.c:79: first defined here
[... cut ...]
/usr/bin/ld: opencl_tc_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/opencl_tc_fmt_plug.c:90: multiple definition of `psalt'; truecrypt_fmt_plug.o:<REDACTED>/FACT_core/src/plugins/analysis/users_and_passwords/bin/john/src/truecrypt_fmt_plug.c:127: first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1724: ../run/john] Error 1
make: *** [Makefile:190: default] Error 2
```

It's huge, I know but this is everything I can see. I hope you can make sense of this.

jstucke, 2021-02-25:
If you can build it separately, then I'm not sure what's wrong. It could possibly be a problem with compiling concurrently. What happens if you remove the ` -sj4` from line 45 in `src/plugins/analysis/users_and_passwords/install.sh`? Did you compile with `--disable-openmp`?

jstucke, 2021-02-25:
Maybe it's related to OpenCL. Does it compile if you add `--disable-opencl` to the ./configure step?

rhelmke, 2021-11-12:
Closing due to both inactivity and missing reproducibility.

closed 2021-11-12 by rhelmke