fkie-cad/FACT_core #548: Integration with LVFS and linux firmware update service?

sei-vsarvepalli, 2021-02-26:
Hello researchers,

Thank you for this great work. Is there any plans to integrate this work with LVFS https://fwupd.org , where firmware updates are being put in regularly by vendors?  The firmware.xml has a schema that richly represents each firmware file which can be used to inform FACT of the firmware and choose the right analysis and the type of features of the firmware.

--- snip sample of a "release" in lvfs firmware.xml  --- 
    <releases>
      <release version="67072" timestamp="1606479307" urgency="high">
        <location>https://fwupd.org/downloads/44a4cdd03322f3504f0e822d1ea19c7a64fc477afda0e2f67dcd01daad556f19-firmware.cab</location>
        <checksum type="sha1" filename="44a4cdd03322f3504f0e822d1ea19c7a64fc477afda0e2f67dcd01daad556f19-firmware.cab" target="container">5bdf69363a8bd45ea8b942022045bf2a8008d7ba</checksum>
        <checksum type="sha256" filename="44a4cdd03322f3504f0e822d1ea19c7a64fc477afda0e2f67dcd01daad556f19-firmware.cab" target="container">da1cff7fde0721537ff00f08b0dfd6edbad24f38be4568d36e3099556df7cc33</checksum>
        <checksum type="sha1" filename="firmware.bin" target="content">3b852921275938706796bdfe1481acaafeea9586</checksum>
        <checksum type="sha256" filename="firmware.bin" target="content">672ab1683fb40c6a0f006e8155c968c7dab0182c7fe4f580e0713fa0b2adf629</checksum>
        <description>
          <p>Dell highly recommends applying this update as soon as possible. The update contains changes to improve the reliability and availability of your Dell system.</p>
          <p>Security issues fixed:</p>
          <ul>
            <li>CVE-2020-8694</li>
            <li>CVE-2020-12356</li>
            <li>CVE-2020-12355</li>
            <li>CVE-2020-12303</li>
            <li>CVE-2020-8752</li>
            <li>CVE-2020-8749</li>
            <li>CVE-2020-8746</li>
            <li>CVE-2020-8755</li>
            <li>CVE-2020-8747</li>
            <li>CVE-2020-8754</li>
            <li>CVE-2020-8760</li>
            <li>CVE-2020-8756</li>
            <li>CVE-2020-8757</li>
            <li>CVE-2020-8705</li>
            <li>CVE-2020-8745</li>
            <li>CVE-2020-8744</li>
            <li>CVE-2020-8753</li>
          </ul>
        </description>
        <size type="installed">17682822</size>
        <size type="download">17695195</size>
        <artifacts>
          <artifact>
            <location>https://fwupd.org/downloads/44a4cdd03322f3504f0e822d1ea19c7a64fc477afda0e2f67dcd01daad556f19-firmware.cab</location>
            <filename>firmware.cab</filename>
            <checksum type="sha1">5bdf69363a8bd45ea8b942022045bf2a8008d7ba</checksum>
            <checksum type="sha256">da1cff7fde0721537ff00f08b0dfd6edbad24f38be4568d36e3099556df7cc33</checksum>
            <size type="installed">17682822</size>
            <size type="download">17695195</size>
          </artifact>
        </artifacts>
      </release>

Vijay

jstucke, 2021-03-01:
Hi, there are currently no plans to integrate this into FACT directly. It should be easy to write a script, that uploads the firmware automatically via the [Rest API](https://github.com/fkie-cad/FACT_core/wiki/Rest-API), though.

sei-vsarvepalli, 2021-03-02:
Thanks @jstucke 

Yes - we are already are trying this in an automated cycle with the current API. There were a number of firmware with vulnerabilities were not matched in the "cve lookup".   The LVFS data has this type of information which you can see in my earlier example.  In the example Dell firmware highlights the CVE's that were addressed by a particular version of "firmware"  As the current code depends primarily on CPE lookup for CVE, this would be a good way to augment .

I would be happy to collaborate with others interested in this work so we can take such work forward.

Vijay

jstucke, 2021-03-03:
The way the CVE lookup works in FACT is as follows: the "software_components" plugin tries to match software with signatures. The "cve_lookup" plugin then tries to find the appropriate CPE entries that match the software. Once you have the CPE entries it is then trivial to look up the matching CVE entries.

For this to work in your case, there would have to be some signatures for the respective software components (e.g. Intel AMT or CSME) including the version (or some other way to detect them). Then the rest should work automatically. Writing such signatures may be difficult or even impossible, though.