fkie-cad/FACT_core #554: Installation on RedHat Enterprise Linux release 8.3

Key2Kidom, 2021-03-23:
I get error FACT installer 1.1 distribution ( rhel 8.3) is not supported. FACT installer requires Ubuntu. this error shows up during src/install.py command. I am running Red Hat Enterprise Linux release 8.3

jstucke, 2021-03-24:
Hi, RHEL is not officially supported but the Fedora installation could still work. To try it, you need to change line 131 in `src/install.py` from 
```python
    if distro.id() == 'fedora':
```
to 
```python
    if distro.id() in ['fedora', 'rhel']:
```

Key2Kidom, 2021-03-30:
I did update     if distro.id() == 'fedora': to  ['fedora', 'rhel']: the issue. it seem to be removing the python package when I run src/install.py.  python3 installation is successful but after this command src/install.py it uninstall the python package

jstucke, 2021-03-31:
Only the packages
- python3-pip
- python3-setuptools
- python3-wheel
are removed during the installation in order to install the latest versions the "official" way using get-pip.py
If this doesn't work in your case, you could try reinstalling the packages.

Key2Kidom, 2021-04-21:
Had to change my distro. thanks @jstucke

closed 2021-04-21 by Key2Kidom